LibreHardwareMonitor/LibreHardwareMonitor #5: Is it possible to implement fan speed control in this application?

VioletGiraffe, 2017-09-11:
A question to those who understand how this application works, what it does, and what Nuvoton chips do. If I have a Nuvoton chip on my motherboard, is it possible to expand this application with fan control features for it, or does that require some entirely different code / protocols? Since Speedfan is abandoned, it's high time to create an alternative that works on modern motherboards, and I'd like to start working on it, but it would helpful if someone confirms that is indeed possible.

Phyxion, 2017-09-12:
Yes, it's certainly possible. Hardware deriving from IControl (https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/Hardware/IControl.cs) or implementing it can be used to specify the fan speed.

VioletGiraffe, 2017-09-12:
Thank you very much! I'm excited about contributing to this project, starting with adding support for my NCT6795D chip.

Phyxion, 2017-09-12:
A generic implementation of NCT6795D already exists in this fork 👍

Eno77, 2017-12-11:
Fan Control is already implemented and Working. (right click on the Fan Control and choose the last menu item: Control -> Manual)

VioletGiraffe, 2017-12-11:
@Eno77: yes, thanks. It took me a while to find it (or, rather, notice). 
There is a bug, though: sometimes (rarely) the change in fan percentage is reflected in the UI but does not actually apply.

Revertron, 2018-06-14:
It would be great to implement automatic fan control according to CPU temperature.

VioletGiraffe, 2018-06-14:
@Revertron  According to any desired temperature, for that matter. It's not difficult controlling CPU fan versus CPU temp - most modern motherboards let you define a curve in UEFI setup. But liquid cooling people need some more flexibility, and that's where modern software and UEFI falls short. Speedfan could do it all, but it's no longer maintain and lacks support for modern NCT chips, sadly.

giofrida, 2019-08-04:
I also need a fan controller. On my old Asus P8Z68 motherboard I used to use Speedfan to control case fans depending on GPU termperature, but now that I have upgraded to Ryzen (Asus X370 Prime Pro, IT8665e) Speedfan no longer works. Also, current OHM version does not show any fan speed data from IT8665e, apart from the CPU fan.

Phyxion, 2019-12-17:
Closing this, if anything is broken please submit a separate report for it.

closed 2019-12-17 by Phyxion