LibreHardwareMonitor/LibreHardwareMonitor #6: The plot doesn't look right, possibly due to display scaling

VioletGiraffe, 2017-09-12:
[image]

This looks wrong, aside from the fact that nothing is actually being plotted (but maybe I just don't understand how to use this feature). This on a 4K monitor with 175% scaling; it could be that the scaling is the reason for the truncated "0:00" label. Just a guess.

Maddimax, 2017-09-17:
Once you have opened the Plot, there are checkboxes next to each Sensor in the main menu. Check those to get stuff into the Plot.

Maddimax, 2017-09-17:
I cannot reproduce the issue with the truncated 0:00. ( I tried 125%, 150% and 175% scaling ). 
Can you check whether the plot is also cropped if you choose on of the other options under "Menu=>Options=>Plot Location"

VioletGiraffe, 2017-09-17:
Yes, it is truncated with all those options.

closed 2018-01-05 by Phyxion