LibreHardwareMonitor/LibreHardwareMonitor #11: Values in the tray icon: the font color has to be customizable

VioletGiraffe, 2017-09-17:
Since the task bars are most often of dark color, the black readings are unreadable. Example:

[image]

The text color has to be customizable; at the very least a choice between black and white would suffice, and white should be the default.

Maddimax, 2017-09-17:
You can right click the tray icon and select "Change color" to change the color ;)

VioletGiraffe, 2017-09-17:
Hmm. That's good, thanks. A bit counter-intuitive as I was looking for it in the application settings.
Still, wouldn't white be a more logical choice for the default color?

Maddimax, 2017-09-17:
I would agree since a black taskbar is the default on Windows 10. Maybe it should default based on the windows version.

VioletGiraffe, 2017-09-17:
The last desktop Windows OS with the light-colored default task bar was Win 2k. It's pretty safe to disregard that possibility, in my opinion :)

closed 2017-09-19 by Phyxion