LibreHardwareMonitor/LibreHardwareMonitor #12: New releases

Maddimax, 2017-09-17:
What do the maintainers think about creating releases in this fork? The last one (0.8) is quite dated already.

Phyxion, 2017-09-18:
That's certainly something that should be done. In the meantime, the build artifacts are always available here:
https://ci.appveyor.com/project/LibreHardwareMonitor/librehardwaremonitor/build/artifacts

Eno77, 2017-12-08:
There seem not to be any artifacts because build error?

Phyxion, 2017-12-08:
Yes, it won't build, I already tried to fix it, but I'm no appveyor guru and I'm not sure how to fix it. Any help is welcome 👍

Eno77, 2017-12-09:
1.0.24 is the last one which builds.
next one adds a new project to the solution (the lib *.csproj) - which seems to be the reason because appveyor does not realize the solution.

appveyor.yml points to the *.sln file - is it using this yml file, or maybe errouneously trying to use the .travis.yml ? (just guessing, im no appveyor guru too)

Eno77, 2018-01-16:
any news for the appveyor build? - btw. I am really looking forward to the nic support.

Phyxion, 2018-01-27:
Build is finally passing again, took some work but it seems to work now 👍

closed 2018-01-27 by Phyxion