LibreHardwareMonitor/LibreHardwareMonitor #20: Support for netstandard

LORDofDOOM, 2017-11-13:
I created already have a working version::

[https://github.com/LORDofDOOM/LibreHardwareMonitor/tree/netstandard](https://github.com/LORDofDOOM/LibreHardwareMonitor/tree/netstandardhttps://github.com/LORDofDOOM/LibreHardwareMonitor/tree/netstandard)

I have changed nvidia API to use dll without PInvokeDelegateFactory wrapper, this is needed because DefinePInvokeMethod is not supported in netstandard (line 64 in PInvokeDelegateFactory).

I drop ATI and TBalancer for now because they also need PInvokeDelegateFactory and I don't own these hardware to rewrite this method and test. Possibly someone own ATI Card and can change/test this. Should be simple to fix - See my change for nvidia :-)

Phyxion, 2017-11-13:
If AMD uses similar dll exports it should work like this as well. I have AMD hardware to test this with, so if you want to change that as well then I'll test it for you.

Regarding .NET Standard, I think that we can just support .NET Standard 1.0, which is .NET Framework 4.5 and use it as the default project. I don't think we need to support .NET 3.5 (which is the current framework) any longer (anyone can update freely), and I also don't see any benefit in .NET 4.0 support apart from it supporting Windows XP. But since XP isn't supported for 3.5 years now I don't think we should support that any longer. I'd like some more feedback on this though.

LORDofDOOM, 2017-11-13:
I've rewrite ATI to use dll -> Tested on my notebook - working. 

Please check if it works for you too

LORDofDOOM, 2017-11-13:
I've created a clean mergeable branch for netstandard - See my pull request :-)

Phyxion, 2017-11-14:
Thanks 👍 
I'll check it out this weekend (maybe earlier if I can find some extra time somewhere). I'm personally in favor of changing to .NET 4.5(maybe .2) as this version is most likely already installed on most systems.

Phyxion, 2018-01-28:
Is merged now, so closing.

closed 2018-01-28 by Phyxion