LibreHardwareMonitor/LibreHardwareMonitor #33: Installing driver "C:\LibreHardwareMonitor\OpenHardwareMonitorLib.sys" failed and file exists

jtcmedia, 2018-02-25:
Several sensor values are missing when executing LibreHardwareMonitor even when Run as Administrator. Report.txt states:
```
Ring0

Status: Installing driver "C:\LibreHardwareMonitor\OpenHardwareMonitorLib.sys" failed and file exists
First Exception: Service already exists
Second Exception: StartService returned the error: This driver has been blocked from loading (Exception from HRESULT: 0x800704FB)
```
This is on a ASUS ROG MAXIMUS X HERO motherboard.

Fungusware, 2019-09-02:
Hey, same issue here. The odd thing is that the OpenHardwareMonitor UI does not do this even on the same machine, its only when I use the OpenhardwareMonitorLib in my own software that this occurs.

Same machine, same Library, different UI app.

Fungusware, 2019-09-02:
Wow, just found the solution, My UI that was using the Library was set to 'AnyCPU' and "Prefer 32Bit' so that the wrong sys driver was trying to be loaded. Now its all good.

nick-lee, 2021-08-21:
I got same error these days, actually the command "OpenHardwareMonitorReport.exe" need run with administrator privilege , otherwise, it will report this error message. if you run on "cmd" shell, you need run this shell as administrator.

closed 2022-09-23 by PhyxionNL