LibreHardwareMonitor/LibreHardwareMonitor #42: Tdie is reported same as Tctl for Ryzen 2700X

SorceressEpele, 2018-06-16:
The Tdie figure is supposed to be the actual CPU Temp.
Tctl is a control temperature that is used for reporting to the cpu fan for cooling purposes, and thus is always higher than the Tdie figure.

Tdie is reported correcting in HWMonitor (as Temperature: Package (Node 0)) and HWiNFO in comparison.

SorceressEpele, 2018-06-16:
Looking through the code, it just appears there is no offset defined. The offset for the 2700X is 10.0 (like the 1910/1920)

Phyxion, 2018-06-19:
If you could submit a PR for the 2700X then it'd be appreciated. I don't have one available to test 👍

SorceressEpele, 2018-06-20:
After performing a BIOS update, the temperature value has also changed completely. It's now reporting completely the wrong value.
HWMonitor reports it at a correct 29°C, whereas I'm getting a Tctl figure of 88.5°C now.

Phyxion, 2018-07-09:
Is the latest build now working for you? There's been some 2700X fixes.

SorceressEpele, 2018-07-09:
Latest version fixes this issue. :D

closed 2018-07-09 by SorceressEpele