LibreHardwareMonitor/LibreHardwareMonitor #56: Samsung 960 EVO shown as Generic Hard Disk

Erik9519, 2018-09-02:
I have a Samsung 960 EVO and it shows up as a Generic Hard Disk and only has the metric Used Space.

selfjc, 2019-01-02:
It appears that when LibreHM was forked from OpenHM that they missed [#PR984](https://github.com/openhardwaremonitor/openhardwaremonitor/pull/984) that added the nvme SSD support for Samsung SSDs or it was broken during some updates since ~June 2017.

I forked the above and merged in [#PR884](https://github.com/openhardwaremonitor/openhardwaremonitor/pull/884) specific for my ASUS Z-370-E Gaming motherboard that uses the Nuvoton NCT6793D chip. I think voltage #11 is one-half of the DRAM voltage.

Since this LibreHM has been rewritten so much, I don't know where to begin adding in the nvme SSD functionality. =(

Here is my report with the working PR948 and PR884 from this repo [OtherOpenHM](https://github.com/selfjc/openhardwaremonitor):

[OpenHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/2719722/OpenHardwareMonitor.Report.txt)

fredriklundholm, 2019-08-16:
Is there a location where a binary version of librehardwaremonitor  can be downloaded with this fix in place?
Tried the latest release linked from the front page but no dice.  (Release-zip)

selfjc, 2019-08-17:
Hi @fredriklundholm , try this compiled version I made on Windows 10 for my Asus Z-370-E-Gaming motherboard with a Samsung 970 Evo SSD. 

[Open Hardware Forked Zipped - Dropbox Link](https://www.dropbox.com/preview/Public/OpenHardwareMonitor.zip?role=personal)

This should be an exact compile from here:
[selfjc/openhardwaremonitor](https://github.com/selfjc/openhardwaremonitor)

fredriklundholm, 2019-08-18:
Hi @selfjc  thanks for your reply. I could not find anything in your dropbox link. It gives an error "The folder ‘/Public/OpenHardwareMonitor.zi...’ doesn’t exist." 
If you have a chance tp revisit the data I would really appropriate it.

Phyxion, 2019-10-11:
NVMe support has been added, closing...

closed 2019-10-11 by Phyxion