LibreHardwareMonitor/LibreHardwareMonitor #77: Tutorial required

MIFCOM-TD, 2019-01-07:
Hello there,

I'm a software developer a MIFCOM GmbH, a High-End system builder company from Germany. I'm working right now on integration of LibreHardwareMonitor directly into our internal PC testing suite. I've already made some changes to forked version of this repository, like updated .NET version and added some utility functions like GetGPUDriverVersion(). Those will be probably kept private as I'm not sure anyone beside me will need them.

But, we, as a commercial OEM system builder have access to many different hardware pieces. Most of them are brand new and few of them are on our table even before official release. I could get my hand early on new mainboards and CPUs, add support to them into LibreHardwareMonitor and create pull request with the changes. 

Problem is, that I think I know how to add support for new CPUs, but I'm lost when it comes to new mainboards. Right now I've got at least 3 different mainboards that have no sensors at all in OHM. Maybe some of you could give me some tutorial or answer a few question, so I could implement support for those mainboards and in return I would create a pull request for each new mainboard I will add. This could really help this project.

You could write me directly at t.dylewski@mifcom.de

Thanks!

w-michal, 2019-01-09:
Most likely those mainboards have a new/uncommon chip that is used for sensors. Known chips are in LPC\Chip.cs.

First, you have to figure out what kind of chip given mainboard have. This should be part of a mainboard spec. If you can't find that info, then you can guestimate the chip by adding logging/break point inside DetectWinbondFintek and DetectIT87 functions and investigating what kind of Chip ID is reported on LPC ports. 

Then, you have add support for that chip. If you are lucky, the unknown chip is just a new revision of well known chip series (like NCT67XX) and you just need to add new chip ID to support (See https://github.com/openhardwaremonitor/openhardwaremonitor/pull/1005 as an example). But if that is not the case, then you have to deep dive into low-level spec of a given chip (example http://www.nuvoton.com/resource-files/NCT6796D_Datasheet_V0_6.pdf ) and how it communicates through LPC ports. Usually, the code that communicates with given already exist somewhere (for example, in linux https://github.com/torvalds/linux/blob/master/drivers/hwmon/nct6775.c ) and you would have to map the basic sensor reading behavior into new LPC chip implementation.

MIFCOM-TD, 2019-01-09:
Thanks, that was very helpful. I've created pull request with new Nuvoton chip. Please check if it's looking fine.

I've also got Clevo Notebook with IT8587 chip (0x8587), but I cannot make it to show sensors the same way using code 0x87XX chips. Any idea why?

w-michal, 2019-01-10:
Huh. Apparently Clevo did something custom (IT85XX does not appear to be "super io" chip like IT87XX). In this case you will have to implement the support for Clevo on your own. Looks like Clevo mainboards uses EC / ACPI for sensors and fan control. Some code here:
https://github.com/tuxedocomputers/tuxedo-fan-control/blob/master/native/ec_access.cc
Also, interesting thread with hwinfo author:
https://www.hwinfo.com/forum/Thread-Solved-Monitor-CPU-and-GPU-fan-RPM-on-Clevo-laptops

MIFCOM-TD, 2019-01-10:
Yeah, from what I know about Clevo, they have everything on a single chip - fans, temperatures and even RGB back-light for keyboard. 
But knowing how much issues there is with their drivers, I think I will simply forget about adding Clevo support. 

Have you checked my pull request? Is it looking OK?

w-michal, 2019-01-10:
PR looks good, but I don't have write access to LHM repo to merge it. You have to wait for @Phyxion

Phyxion, 2019-01-11:
PRs are welcome @MIFCOM-TD , including additional information such as driver versions for GPU. 

Sorry for taking so long to respond to PRs here, for some reason GitHub decided not to notify me of anything anymore :\

closed 2019-01-14 by MIFCOM-TD