LibreHardwareMonitor/LibreHardwareMonitor #99: Code Style Changes

aquacomputer, 2019-07-09:
For future development I would like to change the code style of the original OHM project and control it with the file .editorconfig.
The original project is now no longer active and it makes no sense from my point of view to stick to the rather broken code style.

Since there are few active forks, I think this is feasible.

[editorconfig.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/3372915/editorconfig.txt)

Phyxion, 2019-09-09:
I initially was hesitant to make code style changes in order to make it possible to merge back in main. But I don't think that'll ever happen and there are too many changes anyway so I agree this is a good idea now. I would opt to change it to the "standard" C# code style. Pretty much everything will need to be changed then.

sebastian-dev, 2019-09-12:
This is the default code style config.
[editorconfig.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/3603918/editorconfig.zip)
With this config the most ste settige and namig conventions covered.
Whenever a file is canged, we have to autoformat the file and then merge.

Phyxion, 2019-09-13:
Sounds good, little feedback.
- There are now many "this." unnecessary and they can be removed.
- In editor config, most are changed to error instead of suggestion. For initial convert this is OK, but I think it should be a suggestion - or warning - as in some cases it's not an error. One case is below.
- The WindowPos structure has now renamed fields. Native types should keep their original casing when possible. The struct is already named incorrectly as WindowPos instead of [WINDOWPOS](https://docs.microsoft.com/en-us/windows/win32/api/winuser/ns-winuser-windowpos) so that might need to change anyway.
- Statics shouldn't be prefixed with s_ (see above). I think the constant settings can be mostly copied.
- Some properties have a backing field that can be converted to an auto property. I'm not sure if VS can convert that automatically, but ReSharper can (I can make those changes afterwards).
- Once the GUI part is done, I think we should try to merge that in main first, afterwards the library can be done.

sebastian-dev, 2019-09-13:
Check the project for merging the changes.
The most "this." are removed the d_ rule is removed, the static behavior is now the same as normal public and private fields.

Phyxion, 2019-09-13:
I wanted create a PR for this to compare changes but it seems you didn't base it on the last commit in master. Can you rebase to master? I can then create a PR for it and check things off 👍

sebastian-dev, 2019-09-13:
this is from the last master version.

Phyxion, 2019-09-13:
Yes, but not based from the last commit (https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/d92d88d905087dc797bf6625ad85b54b776c4210). There are many unrelated commits now (outside of the GUI).

sebastian-dev, 2019-09-13:
https://github.com/aquacomputer/LibreHardwareMonitor/tree/codestyle/Hardware
all your changes are up to date

Phyxion, 2019-09-13:
I see, it's a bit confusing that a lot of earlier commits are also listed. I also saw changes to WMI/Utilities but I guess those are changed by editorconfig as well. I'll comment on the PR now.

Phyxion, 2019-09-13:
First part is merged now. Lib can be done in another PR reforked from master 👍

sebastian-dev, 2019-09-13:
https://github.com/aquacomputer/LibreHardwareMonitor/tree/codestyle
during the refactoring i move all win32 functions to the interop folder/classes

sebastian-dev, 2019-09-14:
New Project structure with 90% refactoring is online

Phyxion, 2019-09-23:
Much better structure and readability now @sebastian-dev, I've cleaned it up a little and merged.

commit 73378f7c (2019-09-23): Code style #130 (#132)

This closes #99.
closed 2019-09-23 by Phyxion