LibreHardwareMonitor/LibreHardwareMonitor #110: latest release on github 'release' page is stuck at 0.8 => creates false impression of 'dead' project

mi-hol, 2019-07-23:
Latest release directly downloadable from GitHub is from 6 Nov 2016 version 0.8.0 !

As a result GitHub feature 'watch releases' doesn't alert potential users and creates the false impression of a 'dead' project.

Phyxion, 2019-10-10:
I have removed all releases from GitHub and will try to get AppVeyor working again so executables can be downloaded again.

closed 2019-10-10 by Phyxion

mi-hol, 2019-11-06:
@Phyxion it seems the issue is still open. How about reopening it?

Phyxion, 2019-11-06:
No, there are no more releases listed. Latest builds are done through AppVeyor and linked on main page as such.

Phyxion, 2019-11-06:
There are no more releases listed, so this issue is no longer valid. AppVeyor is working again so the latest builds can be downloaded through it (and they're linked on the main page as such).