LibreHardwareMonitor/LibreHardwareMonitor #116: Dark theme

rrhuffy, 2019-08-13:
What about adding a dark theme?
So instead of this:
[image]
It can look like this:
[image]

123Inge, 2019-10-08:
This is needed. Please

KyodaiKen, 2019-10-12:
I am working on a HTML5 based chart dashboard for the plots in #148 but it now tempts me to design a new UI using WPF... But the time ... :C

123Inge, 2019-10-17:
I might be the only one that does not care about the plotting in dark mode. Only want the 'main' window in dark mode.

KyodaiKen, 2019-10-20:
As I said it is possible to do this by changing the UI to WPF and then using themes. I wonder what the main developers of LHM think about this idea.

rvleic01, 2020-04-18:
Please add a dark mode, will make this so much better to have up on a second screen

hl2guide, 2020-05-23:
Gotta second that! 👍

xRayUnlimited, 2022-01-05:
Does anyone know where the colors are contained? I'm looking at UI> PlotPanel.cs, which is the only place I've found Rgb color values referenced. Once I find the objects, I should be able to theme it/change the colors.

xcasxcursex, 2022-01-05:
> Does anyone know where the colors are contained? I'm looking at UI> PlotPanel.cs, which is the only place I've found Rgb color values referenced. Once I find the objects, I should be able to theme it/change the colors.

Nowhere in particular: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/search?q=color

MRog40, 2022-10-18:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/847

Tamarindo94, 2023-02-01:
there is need

314159DD, 2023-04-28:
is there an update on this? Is it planned to implement dark mode? 
Makes it more eye friendly to be left on a second monitor at night.
Please implement a dark theme.

KyodaiKen, 2023-04-28:
I have moved on to HWiNFO because my last pull request was modified and has been broken (It was support for high core count Threadrippers). I am also running a 13900KS now. I now use remotehwinfo to plot graphs on a laptop screen on the wall using a Raspberry Pi 4. So, unfortunately there will be no HTML5 dashboard for LHM anymore from my side. But it's very scuffed. When I find time I can put this onto Github as an addon to HWiNFO.

camoody1, 2024-07-29:
JayzTwoCents requested a Dark Mode today on his latest video where he talks about your software.
https://www.youtube.com/watch?v=sX2ThUWNXio

progenitor-amborella, 2024-07-29:
> JayzTwoCents requested a Dark Mode today on his latest video where he talks about your software.
> 
> https://www.youtube.com/watch?v=sX2ThUWNXio

Indeed, the issues J2¢ mentioned should get posted / bumped to avoid duplication issues, as people that don't understand much about git will likely flood this repo. 

Great to have them! But certainly would like to make it easier on the maintainers.

closed 2025-09-27 by PhyxionNL