LibreHardwareMonitor/LibreHardwareMonitor #121: No nvidia powerUsage with openhardwaremonitorLib.dll

SaulGoodman1337, 2019-09-03:
Hello, everyone,
please bear with me, C# has been a complete stranger to me so far.

I currently have a project in mind which is about controlling my housing fans with the powerUsage of my CPU and GPU.

first i tried something with powershell, but now i ended up at a C# project where i imported the openhardwaremonitorLib.dll.

I can do everything so far:
- get power CPU in watt
- Control Case fan

What I still need is the powerUsage my Nvidia GPU. In openhardwaremonitor this is also shown to me. But I don't get anything like that in C#. The variable powerUsage is zero.

I strongly suspect that the NVML needs to be initialized, don't I? if I understand the code correctly it will be done sometime at the soundtime of the OpenhardwareMonitor GUI.... but unfortunately I don't know exactly.

Can somebody help me to get the watt consumption of my GPU out?

If necessary I could get the value from the WMI object for the time being, but of course it is not clean.

[image]

SaulGoodman1337, 2019-09-05:
can nobody help me ? .... :(

Phyxion, 2019-09-09:
You will need to update the sensors first, create a loop that updates the hardware every second for example.

HUNTERok, 2021-06-02:
@SaulGoodman1337 @karthik101987 
Switching target platform to x64 in your project settings resolve this issue.