LibreHardwareMonitor/LibreHardwareMonitor #133: appveyor

sebastian-dev, 2019-09-23:
https://ci.appveyor.com/project/LibreHardwareMonitor/librehardwaremonitor/build/artifacts
is not working with the new project structure maybe appveyor.yml need some fixes

chtof, 2019-09-25:
It should explain why I do not longer find `bin%2FRelease.zip` on https://github.com/LibreHardwareMonitor/LibreHardwareMonitor.

fboldog, 2019-09-29:
Interesting, build from my last development branch had a proper zip:
https://ci.appveyor.com/project/fboldog/librehardwaremonitor/build/artifacts

Eno77, 2019-10-01:
Console says:

```
Collecting artifacts...
No artifacts found matching 'Bin\Release' path
```

Perhaps the zip file is not in correct path anymore or something?

Phyxion, 2019-10-10:
Took some time and testing but it's creating a release again.

Phyxion, 2019-10-10:
Took some time and testing but it's uploading artifacts again.

closed 2019-10-10 by Phyxion