LibreHardwareMonitor/LibreHardwareMonitor #134: Nothing displayed for AMD RX 5700 XT GPU

binary100100, 2019-09-23:
It only shows the card but no details.
Latest Windows 10, latest drivers.
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/3644229/LibreHardwareMonitor.Report.txt)

fboldog, 2019-10-09:
Hi,

could you try this version? 
https://ci.appveyor.com/project/fboldog/librehardwaremonitor/build/artifacts
It's use some newer ATI api call.

Is there any information displayed about your card?

KyodaiKen, 2019-10-12:
I know someone with this exact GPU (AiB partner version, not reference), I asked him if he can help us troubleshooting. According to him, he has seen data from this card once and it was gone later with the same LHM version. I will keep y'all updated.

binary100100, 2019-10-20:
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/3748036/LibreHardwareMonitor.Report.txt)

dONKdONKdONKdONK, 2019-10-22:
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/3754347/LibreHardwareMonitor.Report.txt)

TrickyDee42, 2019-10-28:
Any news on this?

fboldog, 2019-10-31:
Hi, could you run my small tester app? @dONKdONKdONKdONK , @TrickyDee42 , @binary100100 
https://github.com/fboldog/adl-tester/suites/290283753/artifacts/219541
And post the output to this issue?

It runs from command line, without any admin rights and actually opens the first adapter. 
For example: my Vega56 produces this output:
[image]

Of course you can check the source code:
https://github.com/fboldog/adl-tester

dONKdONKdONKdONK, 2019-11-01:
[image]

fboldog, 2019-11-01:
@dONKdONKdONKdONK thanks.
Thanks, just as I excepted. :(
The 5700 (XT) cards using the new v8 aka. OverdriveN api. 
I will create a new test app for this in the coming days.

fboldog, 2019-11-01:
@dONKdONKdONKdONK one more question.
does your your 5700xt is supported by GPU-Z and does sensors tab are filled with data?

dONKdONKdONKdONK, 2019-11-01:
[image]

cednieb, 2019-12-02:
fboldog , some progress ?

fboldog, 2019-12-03:
Hey guys, so, actually not really.
AMD isn't update their framework (AMD Display Library aka. ADL) since months. Which is not really supports NAVI GPUs. I tired many ways, without any real progress. :(
Actually I using a Vega56 card, debugging without a 5700 (XT) is hard or nearly impossible. :D

cednieb, 2019-12-03:
It seems that works \display-library-master\Sample\Overdrive8 at less for the hotspot

-----------------------------------------
Adapter Index[0]
 -----------------------------------------
itemID:OD8_OPERATING_TEMP_MAX: is not Visible
OD8_OPERATING_TEMP_MAX:maxValue:110
minValue:50
defaultValue:110
-----------------------------------------
itemID:OD8_FAN_TARGET_TEMP: is not Visible
OD8_FAN_TARGET_TEMP:maxValue:100
minValue:25
defaultValue:75
-----------------------------------------
itemID:OD8_POWER_PERCENTAGE: is Visible
actualValue:0
OD8_POWER_PERCENTAGE:maxValue:20
minValue:-50
defaultValue:0
-----------------------------------------
ADLSensorType: PMLOG_TEMPERATURE_EDGE - Current Temp
PMLOG_TEMPERATURE_EDGE.supported:1
PMLOG_TEMPERATURE_EDGE.value:35
ADLSensorType: PMLOG_TEMPERATURE_HOTSPOT - Junction Temp
PMLOG_TEMPERATURE_HOTSPOT.supported:1
PMLOG_TEMPERATURE_HOTSPOT.value:35
-----------------------------------------

cednieb, 2019-12-03:
Thats works
[https://pastebin.com/99QmJ5TU](url)

fboldog, 2019-12-03:
Hm, i look after it. Thanks for this information.

cednieb, 2019-12-03:
I try to fix your adltester .
Can you help ?

fboldog, 2019-12-04:
I pushed an initial version of this query.
https://github.com/fboldog/adl-tester/suites/342680815/artifacts/552948
Please someone with 5700 (or XT) test the output.

cednieb, 2019-12-04:
https://pastebin.com/hyFukRiv
Source code please

fboldog, 2019-12-04:
Thanks, looks good. Going forward (tomorrow) with checking settings which is enabled, etc.

fboldog, 2019-12-05:
Please someone test again, but now in my LHWM build. :)
https://ci.appveyor.com/project/fboldog/librehardwaremonitor/build/artifacts
Thanks!

cednieb, 2019-12-05:
Source code plz , I can compile it .

fboldog, 2019-12-05:
https://github.com/fboldog/LibreHardwareMonitor/tree/support-rx5700-series-issue-134

cednieb, 2019-12-05:
That works , I retrieve the same values ( Aida , hwinfo )

cednieb, 2019-12-05:
Thanks for your work

fboldog, 2019-12-05:
Could you make a screenshot?
You're welcome. :)

cednieb, 2019-12-05:
http://thisiteisnot.online/ScreenShot001.jpg

dONKdONKdONKdONK, 2019-12-05:
[image]

Running Heaven - GPU Core Clock not reporting correctly.  Everything else just works!
Thanks so much

fboldog, 2019-12-06:
@dONKdONKdONKdONK & @cednieb 
could test the new version (https://ci.appveyor.com/project/fboldog/librehardwaremonitor/build/artifacts - 1.0.21)? 
- fixed voltage display
- added some new values

if it works, this version will be the PR.

dONKdONKdONKdONK, 2019-12-06:
[image]

Shows correct Max Value on core but no SOC value

cednieb, 2019-12-06:
[image]
http://thisiteisnot.online/ScreenShot002.jpg

inks007, 2019-12-07:
@fboldog awesome work, thanks.

[image]

GPU fan shows 65535 RPM / 1323.0% instead of 0 RPM / 0% (it is unplugged).

fboldog, 2019-12-07:
@inks007 disconnected fan fixed
@dONKdONKdONKdONK soc value assigned to bad sensor :) this is fixed too.

Please check the new build at https://ci.appveyor.com/project/fboldog/librehardwaremonitor/build/artifacts (1.0.22)

dONKdONKdONKdONK, 2019-12-08:
That works, thanks again. Now I can keep an eye on my dodgy 5700 xt

inks007, 2019-12-08:
I see that the fan entry is removed (because of the null value) if the sensor returns 65535. Perhaps returning N/A, zero or '-'  might be better. ADL might be reporting -1 for a dead fan, which would be relevant for others. I've modded my 5700XT, so I know that it shouldn't have a fan reading - I can just hide the entries.

My fan header is inaccessible - I can't double check if a connected fan that is stuck/stopped reads 0 RPM or 65535 RPM. Might be useful if @dONKdONKdONKdONK or @cednieb could check this,

dONKdONKdONKdONK, 2019-12-08:
65535 RPM is a driver issue I think. I had it on GPU-Z and Wattman when I first purchased the card. It was resolved by a v-BIOS update and new drivers, it came back with the latest Radeon driver 19.12.1 but a new v-BIOS sorted it out. My RPM now reads either 0 or 400-450 RPM when the fans are stopped.
See pic with GPU-Z, LHW and Wattman OSD
[image]

fboldog, 2019-12-08:
I changed to '-' version, when it's reported as disconnected.
65535 is not a bug, this is the ushort (or UInt16) max value, which is OK for reporting a disconnected fan. :)
New build on same place: https://ci.appveyor.com/project/fboldog/librehardwaremonitor/build/artifacts (1.0.23)

commit 36a9d060 (2019-12-17): Support rx5700 series (#177)

Fixes #134.
closed 2019-12-17 by Phyxion