LibreHardwareMonitor/LibreHardwareMonitor #140: New Intel Pentium Silver CPUs

SearchForTheCode, 2019-10-04:
The new Intel Pentium Silver cpus in notebooks seem to be not working.

I uploaded an report for further analysis. In best case, this is just another detection case, no adjusting to any low level stuff. 

[OpenHardwareMonitor.Report (1).txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/3689181/OpenHardwareMonitor.Report.1.txt)

SearchForTheCode, 2019-11-20:
Any chance for a solution?

hirschmann, 2019-11-21:
Give the latest build a try: https://ci.appveyor.com/project/LibreHardwareMonitor/librehardwaremonitor/build/artifacts

Gemini Lake CPUs are supported now: 404b2a766b99673f7fb811cfc4f45d8df3196d4f

SearchForTheCode, 2019-11-27:
I tried the latest version now, sadly not working for Silver. So the Changelog is true , as now Silver is listed, just Goldmont etc.

If the report does help, please keep me informed!

[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/3896635/LibreHardwareMonitor.Report.txt)

hirschmann, 2019-11-29:
Looks like LHM can't install the Ring0 driver:
```
Ring0

Status: Installing driver "C:\Users\administrator\Desktop\Release\LibreHardwareMonitorLib.sys" failed and file exists
First Exception: Service already exists
Second Exception: Service already exists
```

LHM correctly recognizes your CPU's architecture as Goldmont Plus, but can't read from the model specific registers (MSR) because the Ring0 driver is not installed:
```
Intel CPU

Name: Intel Pentium Silver N5000
Number of Cores: 4
Threads per Core: 1
Timer Frequency: 10 MHz
Time Stamp Counter: Invariant
Estimated Time Stamp Counter Frequency: 1094.54 MHz
Estimated Time Stamp Counter Frequency Error: 0.2014 Mhz
Time Stamp Counter Frequency: 1094.4 MHz

MSR Core #1

 MSR       EDX       EAX

MSR Core #2

 MSR       EDX       EAX

MSR Core #3

 MSR       EDX       EAX

MSR Core #4

 MSR       EDX       EAX

MicroArchitecture: GoldmontPlus
Time Stamp Counter Multiplier: 0
```

Your problem could be caused by other hardware monitoring tools which use the same driver.
Windows Core Isolation could also be a problem.

SearchForTheCode, 2019-11-29:
Thanks for the information :) 
Then i will close this topic and start analysing with core isolation etc.

closed 2019-11-29 by SearchForTheCode