LibreHardwareMonitor/LibreHardwareMonitor #141: AccessViolationException in NVMeSamsung

MoshiMoshi0, 2019-10-07:
Since the big refactoring pr I started getting exceptions when initializing samsung nvme ssd.
I have to let it crash a couple of times and then it works fine.

Also normal ssd sensor data takes very long to update, but again only a couple of times. Then it updates fast.

I guess there is a problem with capturing smart data? I have no idea how to debug/fix this.
Probably broken since [this](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/db51d7f509caba7db9d0721d56d80c9f7972f187) commit.

NVME: Samsung 950 PRO
SSD: Samsung 850 EVO
Win: 10.0.18362.388
LibreHardwareMonitor: Build from [61454e1ae2f2ce18550bbb337719e39a3058884f](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/61454e1ae2f2ce18550bbb337719e39a3058884f)

Exception 1:
```
System.AccessViolationException
   in System.Runtime.InteropServices.Marshal.PtrToStructureHelper(IntPtr, System.Object, Boolean)
   in System.Runtime.InteropServices.Marshal.PtrToStructure(IntPtr, System.Type)
   in System.Runtime.InteropServices.Marshal.PtrToStructure[[LibreHardwareMonitor.Interop.Kernel32+NVME_IDENTIFY_CONTROLLER_DATA, LibreHardwareMonitorLib, Version=0.8.1.0, Culture=neutral, PublicKeyToken=null]](IntPtr)
   in LibreHardwareMonitor.Hardware.Storage.NVMeSamsung.IdentifyController(System.Runtime.InteropServices.SafeHandle, NVME_IDENTIFY_CONTROLLER_DATA ByRef)
   in LibreHardwareMonitor.Hardware.Storage.NVMeSmart.GetInfo()
   in LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.GetDeviceInfo(LibreHardwareMonitor.Hardware.Storage.StorageInfo)
   in LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.CreateInstance(LibreHardwareMonitor.Hardware.Storage.StorageInfo, LibreHardwareMonitor.Hardware.ISettings)
   in LibreHardwareMonitor.Hardware.Storage.AbstractStorage.CreateInstance(System.String, UInt32, UInt64, Int32, LibreHardwareMonitor.Hardware.ISettings)
   in LibreHardwareMonitor.Hardware.Storage.StorageGroup..ctor(LibreHardwareMonitor.Hardware.ISettings)
   in LibreHardwareMonitor.Hardware.Computer.Open()
```

Exception 2:
```
System.ExecutionEngineException
   in System.Runtime.InteropServices.Marshal.PtrToStructureHelper(IntPtr, System.Object, Boolean)
   in System.Runtime.InteropServices.Marshal.PtrToStructure(IntPtr, System.Type)
   in System.Runtime.InteropServices.Marshal.PtrToStructure[[LibreHardwareMonitor.Interop.Kernel32+NVME_IDENTIFY_CONTROLLER_DATA, LibreHardwareMonitorLib, Version=0.8.1.0, Culture=neutral, PublicKeyToken=null]](IntPtr)
   in LibreHardwareMonitor.Hardware.Storage.NVMeSamsung.IdentifyController(System.Runtime.InteropServices.SafeHandle, NVME_IDENTIFY_CONTROLLER_DATA ByRef)
   in LibreHardwareMonitor.Hardware.Storage.NVMeSmart.GetInfo()
   in LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.GetDeviceInfo(LibreHardwareMonitor.Hardware.Storage.StorageInfo)
   in LibreHardwareMonitor.Hardware.Storage.NVMeGeneric.CreateInstance(LibreHardwareMonitor.Hardware.Storage.StorageInfo, LibreHardwareMonitor.Hardware.ISettings)
   in LibreHardwareMonitor.Hardware.Storage.AbstractStorage.CreateInstance(System.String, UInt32, UInt64, Int32, LibreHardwareMonitor.Hardware.ISettings)
   in LibreHardwareMonitor.Hardware.Storage.StorageGroup..ctor(LibreHardwareMonitor.Hardware.ISettings)
   in LibreHardwareMonitor.Hardware.Computer.Open()

sebastian-dev, 2019-10-18:
I have seen this error only 1..2 times during debugging. Not in the Release or without a attached debugger.

MoshiMoshi0, 2019-10-19:
I did some testing with just opening and closing the Computer in a loop and it crashes for me:
* Sometimes on debug/release build in visual studio, on first `Computer.Open`
* Sometimes on debug/release build running manually without visual studio, on first `Computer.Open`
* Rarely when attaching/reattaching the debugger

So for me it can crashes quite frequently, but not consistent. Maybe its a timing issue or an issue with something trying to read smart data at the same time?

MoshiMoshi0, 2019-10-22:
@sebastian-dev I created a PR with a fix, please check if my above reasoning is correct.

PR #160 (2019-10-27): Fix buffer over-read causing AccessViolationException in NVMeSamsung
The data buffer size in `SCSI_PASS_THROUGH_WITH_BUFFERS` is set to only 512 bytes but it is used to create `NVME_IDENTIFY_CONTROLLER_DATA` which is 4096 bytes with `Marshal.PtrToStructure`. This causes buffer over-read.

This could also be solved by creating a new memory block that fits `NVME_IDENTIFY_CONTROLLER_DATA` and copying the smaller `SCSI_PASS_THROUGH_WITH_BUFFERS.DataBuf` there. This is done in [NVMeIntel.cs](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/bae89127b2d6c1813a9b8736a63a5c997541ff9a/L [... cut]
closed 2019-10-27 by Phyxion