LibreHardwareMonitor/LibreHardwareMonitor #164: Add Intel (Integrated) iGPU Support

rkkoszewski, 2019-10-30:
Hi,

It would be great to be able to read information like Temperature, GPU load, Clock, VRAM/Power usage for Intel iGPUs. 

Has anybody attempted to add Intel iGPU Support to Open/LibreHardwareMonitor? Has anybody seen any useful information on this topic? (All I was able to find so far were vague statements that a kernel driver and bit banging might be required). Also Windows 10 shows now the GPU status in the Task manager, is that API somehow accessible?

JaegerFox, 2021-06-18:
only the only open source software that can see gpu usage is process hacker

DjordjeMandic, 2021-06-18:
Im not sure if cpu sockets are routed to have seperate power supply for gpu and cpu, if they are then its up to motherboard, if motherboard is using single vrm for both then you cant do anything expect get whole cpu power usage

ifarbod, 2023-03-23:
> Also Windows 10 shows now the GPU status in the Task manager, is that API somehow accessible?

Yes, it's called [D3DKMT](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmthk/). In fact it's the same API used by Task Manager and [System Informer](https://github.com/winsiderss/systeminformer/blob/d4ba1b2703e7f9b87e84c7a7aff9ee4d75a8ff97/plugins/ExtendedTools/gpudetails.c#L279) You call [this function](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmthk/nf-d3dkmthk-d3dkmtqueryadapterinfo) with [this structure](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmthk/ns-d3dkmthk-_d3dkmt_adapter_perfdata).

ifarbod, 2023-04-04:
I have collected some helpful information regarding the original issue:

LibreHardwareMonitor already supports reporting CPU GT Cores' power (wattage) values by reading the CPU's Module Specific Registers (MSR). so all left is temperature, utilization percent, GPU Clock, and voltage (VID?).

I've read the Intel processor generation datasheets, Intel has a place where all of this info is written, from the BIOS/firmware.

This place is called MCHBAR, to get the base address for it, you'd have to read the bus0, dev0, func0 PCI configuration, offset +0x48, and size 4 bytes.

The address reported back is usually 0xFED10001. Intel code from the Linux kernel does `& ~1` on it, so it's converted to 0xFED10000.

Now that you have the physical memory address, all you have to do is map this to user space virtual memory.

On Linux this is usually done by `mmap`, but on Windows, this process is a little more difficult as you'd need Ring0 access for it.

Luckily, LHM already ships a kernel mode driver, so this shouldn't be an issue, however, we must extend it with memory mapping functionality.

In `DriverDeviceControl`:
```cpp
        case R0IOCTL::MapPhysicalMemory:
            status = MapPhysicalMemory(
                sysBuffer, dic.InputBufferLength, dic.OutputBufferLength, reinterpret_cast<ULONG*>(&len));
            break;
```

```cpp

// Mapped memory information list
using MAPINFO = struct tagMAPINFO
{
    SINGLE_LIST_ENTRY link;
    PMDL pMdl;      // allocated mdl
    PVOID pvk;      // kernel mode virtual address
    PVOID pvu;      // user mode virtual address
    ULONG memSize;  // memory size in bytes
};
using PMAPINFO = MAPINFO*;

SINGLE_LIST_ENTRY lstMapInfo;  // mapped memory information

NTSTATUS MapPhysicalMemory(void* InputBuffer, ULONG InputBufferLength, ULONG OutputBufferLength, ULONG* BytesReturned)
{
    if (InputBufferLength != sizeof(PHYSICAL_MEMORY) || OutputBufferLength != sizeof(PVOID))
    {
        return STATUS_INVALID_PARAMETER;
    }

    auto* physicalMemory = static_cast<PHYSICAL_MEMORY*>(InputBuffer);

    PHYSICAL_ADDRESS phyAddr{};
    phyAddr.QuadPart = static_cast<LONGLONG>(static_cast<ULONGLONG>(physicalMemory->pvAddr));

    // get mapped kernel address
    PVOID pvk = MmMapIoSpace(phyAddr, physicalMemory->dwSize, MmNonCached);

    if (pvk != nullptr)
    {
        // allocate mdl for the mapped kernel address
        PMDL pMdl = IoAllocateMdl(pvk, physicalMemory->dwSize, FALSE, FALSE, nullptr);
        if (pMdl != nullptr)
        {
            PMAPINFO pMapInfo = nullptr;

            // build mdl and map to user space
            MmBuildMdlForNonPagedPool(pMdl);
            // TODO(iFarbod): |MdlMappingNoExecute?
            PVOID pvu = MmMapLockedPagesSpecifyCache(pMdl, UserMode, MmNonCached, nullptr, FALSE, NormalPagePriority);

            // insert mapped infomation to list
            // TODO(iFarbod): replace with ExAllocatePool2,3,WithTag? pick the function based on Windows Version
            pMapInfo = static_cast<PMAPINFO>(ExAllocatePool(NonPagedPool, sizeof(MAPINFO)));

            if (pMapInfo != nullptr)
            {
                pMapInfo->pMdl = pMdl;
                pMapInfo->pvk = pvk;
                pMapInfo->pvu = pvu;
                pMapInfo->memSize = physicalMemory->dwSize;
                PushEntryList(&lstMapInfo, &pMapInfo->link);

                // TODO(iFarbod): copy to outputBuffer?
                RtlCopyMemory(InputBuffer, &pvu, sizeof(PVOID));

                *BytesReturned = sizeof(PVOID);

                return STATUS_SUCCESS;
            }
        }

        // allocate mdl error, unmap the mapped physical memory
        MmUnmapIoSpace(pvk, physicalMemory->dwSize);

        return STATUS_INSUFFICIENT_RESOURCES;
    }
    return STATUS_INSUFFICIENT_RESOURCES;
}
```

[image]

Now, to get the GPU info we want:

- Temperature: `0x5980`, read 32 bits, value is in degrees Celsius with 1°C precision. Only AIDA64, HWInfo64, HwMonitor, and GPU-Z can report this. I've found no known open-source software that has this.
- Engine clock: `0x5948`, also called `GT_PERF_STATUS` in Intel datasheets.

```cpp
constexpr auto GT_PERF_STATUS = 0x5948;
constexpr auto GT_FREQUENCY_MULTIPLIER = 50;
auto frequency = read(GT_PERF_STATUS);

frequency = (frequency & 0xff00) >> 8;
frequency *= GT_FREQUENCY_MULTIPLIER;
```

- VRAM clock: `0x5E00`.

```cpp
MC_BIOS_REQ = 0x5E00

auto frequency = read(MC_BIOS_REQ);
frequency *= 133;
```

I'd love to add them to LHM as a PR but I'm unfamiliar with the codebase and don't have too much free time to experiment with it, nor the EV signing certificate to recompile the OpenLibSys driver. Hopefully, someone else picks this up in the near future.