LibreHardwareMonitor/LibreHardwareMonitor #167: Incorrect temperatures reported on Ryzen 2700X

dunestorm333, 2019-11-14:
[image]

The temperatures reported by LHM are off by ~10C on Ryzen CPUs. My screenshot shows a tool called Sidebar Diagnostics that uses LibreHardwareMonitor as a backend for monitoring in comparison with the official Ryzen Master software by AMD, the screenshot shows a clear disparity between the two reported temperatures.

I have also attached a copy of my system info.
[DxDiag.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/3849004/DxDiag.txt)

TheRadziu, 2019-11-27:
[image]

Same issue with Ryzen 7 3800X. 
Temp in bios matches the temps in Ryzen Master.

Phyxion, 2019-12-04:
What are the values if you compare it with HWiNFO? For me it shows Tctl/Tdie temperature same as Tctl and Tdie in LHM. Ryzen Master reports average temperature (which seems to be the same as HWiNFO's CPU Die (average)).

hirschmann, 2019-12-04:
For my 2700X LHM subtracts the temperature offset as expected:

[image]

@dunestorm333
Maybe the sidebar diagnostics tool uses the T_ctl reading instead of T_die?

@TheRadziu 
T_ctl and T_die are equal on your screenshot, which should be correct for the 3800X.
Afaik AMD gave up the temperature offset crap with Ryzen 3xxx (zen 2).

I'm not sure why LHM is so far off the Ryzen Master temperature readings.
Is it constantly off by ~10 degree? My guess is that @Phyxion is right and it's a average temp. vs actual temp. problem.

ArcadeRenegade, 2019-12-06:
Hi @hirschmann 

In Sidebar Diagnostics I get the CPU temperature like this:

```
ISensor _tempSensor = null;

if (board != null)
{
    _tempSensor = board.Sensors.Where(s => s.SensorType == SensorType.Temperature && s.Name.Contains("CPU")).FirstOrDefault();
}

if (_tempSensor == null)
{
    _tempSensor =
        _hardware.Sensors.Where(s => s.SensorType == SensorType.Temperature && s.Name == "CPU Package").FirstOrDefault() ??
        _hardware.Sensors.Where(s => s.SensorType == SensorType.Temperature).FirstOrDefault();
}
```

From there I just read ISensor.Value.

hirschmann, 2019-12-06:
@ArcadeRenegade
Seems like my guess was right.
LHM only lists 2 sensors for Ryzen: 
- Core (Tctl)
- Core (Tdie)

Therefore in your code, searching for sensors named "CPU" or "CPU Package" fails and it falls back to:
`_hardware.Sensors.Where(s => s.SensorType == SensorType.Temperature).FirstOrDefault();`

The first sensor in the list is Core (Tctl), which will then be shown in the sidebar.

ArcadeRenegade, 2019-12-06:
Got it. I don't have a Ryzen to test with. Should I get where sensor name is equal to "Tdie" or where sensor name is equal to "Core (Tdie)"?

hirschmann, 2019-12-06:
Either Contains("Tdie") or Equals("Core (Tdie)"). You're welcome, glad I could help :)

ArcadeRenegade, 2019-12-07:
Cheers

ArcadeRenegade, 2019-12-07:
Hi @hirschmann 

Another user is reporting the same Tdie temp as the Tctl temp in Libre but their HWInfo and Ryzen Master is showing the Tdie about 10c lower.

[image]

[image]

ArcadeRenegade, 2019-12-07:
Never mind just saw what you posted earlier in the thread.

Phyxion, 2019-12-13:
Please try out the latest build, for some Ryzen CPUs an invalid offset was applied. Keep in mind that LHM reports max and not average, see above for details on that.

ArcadeRenegade, 2019-12-13:
Thanks, @Phyxion . I updated to the latest build.

Unfortunately, I'm getting an error if I try to close an instance of "Computer" class and create a new one and "Open" it. This happens when someone changes their monitoring settings in my application.

```
System.AccessViolationException: 'Attempted to read or write protected memory. This is often an indication that other memory is corrupt.'

   at LibreHardwareMonitor.Interop.NvidiaML.NvmlDeviceGetHandleByIndex(Int32 index)
   at LibreHardwareMonitor.Hardware.Gpu.NvidiaGpu..ctor(Int32 adapterIndex, NvPhysicalGpuHandle handle, Nullable`1 displayHandle, ISettings settings)
   at LibreHardwareMonitor.Hardware.Gpu.NvidiaGroup..ctor(ISettings settings)
   at LibreHardwareMonitor.Hardware.Computer.Open()
   at SidebarDiagnostics.Monitoring.MonitorManager..ctor(MonitorConfig[] config) in C:\Users\ryank\OneDrive\Documents\GitHub\SidebarDiagnostics\SidebarDiagnostics\Monitoring.cs:line 38
   at SidebarDiagnostics.Models.SidebarModel.InitMonitors() in C:\Users\ryank\OneDrive\Documents\GitHub\SidebarDiagnostics\SidebarDiagnostics\SidebarModel.cs:line 100
   at SidebarDiagnostics.Models.SidebarModel..ctor() in C:\Users\ryank\OneDrive\Documents\GitHub\SidebarDiagnostics\SidebarDiagnostics\SidebarModel.cs:line 14
   at SidebarDiagnostics.Sidebar.<<BindModel>b__10_0>d.MoveNext() in C:\Users\ryank\OneDrive\Documents\GitHub\SidebarDiagnostics\SidebarDiagnostics\Sidebar.xaml.cs:line 180
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder.Start[TStateMachine](TStateMachine& stateMachine)
   at SidebarDiagnostics.Sidebar.<BindModel>b__10_0()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot)
   at System.Threading.Tasks.Task.ExecuteEntry(Boolean bPreventDoubleExecution)
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
```

Phyxion, 2019-12-17:
@ArcadeRenegade I cannot reproduce on my own NVIDIA GPU, please try to debug it and submit a PR with a fix.

ArcadeRenegade, 2019-12-17:
Ok if it is just specific to my environment then I will ignore it

Phyxion, 2019-12-17:
It might not be :)

closed 2020-01-17 by Phyxion

schiffy91, 2020-05-12:
I'm seeing a significant disparity between the reported temperature in Sidebar and what I see on AMD Ryzen Master for my 3950x. It's roughly 10-ish degrees. I see the issue was closed, but I don't see that a CL was merged in. @Phyxion and @ArcadeRenegade LMK if you need me to submit something.