LibreHardwareMonitor/LibreHardwareMonitor #169: Automatic fan curve control

pawg, 2019-11-19:
Hi

As speedfan is not updated, I'm in search of a replacement. LibreHardwareMonitor is so great, but fan control is only manual.

Would it be possible to merge this fork into master ?
https://github.com/SorceressEpele/LibreHardwareMonitor

It works wonderfully well, it is elegant, very simple and perfectly integrated into the interface.
This simple addition would make LibreHardwareMonitor a viable and up to date alternative.
Please forgive my GitHub illiteracy, maybe there's a more streamlined way to request something like this.

Thanks

Nama, 2020-01-24:
I have just tested your fork and it works pretty nice.

But it seems, after choosing "New", selecting a sensor and setting the curve, it still expects a sensor to click on. So after clicking somewhere, this error appears. Although I already clicked on a sensor and set the curve already.
[image]

Since you already forked the code and pushed your commits, you can make pull requests. Look for that feature (after updating your code), to use the "streamlined way" :)

PeterSHollander, 2020-02-24:
Hi @pawg or @Nama - Are you planning to submit an updated pull request for this?  Can I help out in any way?  I would love to have automatic fan curve control integrated into Libre Hardware Monitor.  (Might be worth pinging the creator of this fork, [https://github.com/SorceressEpele](https://github.com/SorceressEpele).  I too am new to GitHub, so I'm not quite sure how all of this works!)

Cheers!

Nama, 2020-02-24:
I noticed too late, that pawg is not the owner of the fork^^
The owner needs to do the PR, so maybe we can get his attention ;)
@SorceressEpele

Nama, 2020-03-01:
Found this. This is even based on the newest version of Libre. Sadly, it's not opensource...
https://linustechtips.com/main/topic/1099996-openfancontrol-my-take-on-a-speedfan-replacement/

SorceressEpele, 2020-03-01:
A pull request from my fork cannot be merged automatically. I'm not sure what would need changing to get it to merge successfully, so it might be easier to merge by hand?

The code comes from a pull request on OpenHardwareMonitor that I merged in for personal use.

PeterSHollander, 2020-03-02:
Okay; how can I help on that?  Would it be a matter of focusing on what changes the [original pull request](https://github.com/openhardwaremonitor/openhardwaremonitor/pull/996) made a few years ago to OpenHardwareMonitor, and then re-making those changes under the more recent main LibreHardwareMonitor code?  And are there additional fixes/features that you added in your [fork over to LibreHardwareMonitor](https://github.com/SorceressEpele/LibreHardwareMonitor) that should be taken into consideration?
Thanks for replying to this thread, btw!

pawg, 2020-03-02:
Look here for a very interesting and up-to-date replacement! https://linustechtips.com/main/topic/1099996-openfancontrol-my-take-on-a-speedfan-replacement/

PeterSHollander, 2020-03-02:
Hi @pawg, thank you for the link.  Unfortunately, as @Nama pointed out this project does not appear to be open source.  I have left a post on the forum that you linked to in hopes that the developer may merge it into LibreHardwareMonitor, or at least provide to source code for us to do so ourselves.

SorceressEpele, 2020-03-02:
It's pretty much just focusing on those changes of the original pull request. There's nothing else additional to my fork, outside of a few attempts to poke around to see if I could get the ITE IT8665E to reveal its sensor data, but I didn't have any luck with that.

PeterSHollander, 2020-03-02:
Okay - well great, thanks for the info @SorceressEpele!  I'll see what I can do with it.  Hopefully my first time submitting to GitHub won't be all that complicated to figure out :)

MailYouLater, 2020-03-09:
@PeterSHollander: I saw your post on that LTT thread and the response ("Look into your DMs."), Did @Rem0o get back to you about this?  Any news you can share?

PeterSHollander, 2020-03-09:
@Rem0o did respond, but sadly I have nothing new to share @MailYouLater.  I believe our best bet will be to adapt the [original OpenHardwareMonitor pull request](https://github.com/openhardwaremonitor/openhardwaremonitor/pull/996) to fit into the present fork of LibreHardwareMonitor.
I do have full intention of pursuing this, but it will be at a (very) leisurely pace.

MailYouLater, 2020-03-10:
@Rem0o: I see you've [contributed](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commits?author=Rem0o) to LHM, and comparing your comments in the LTT thread with your contributions, I'm pretty sure you are the same person... so...  Are you planning to open source your "OpenFanControl" project? and assuming (because of the name you chose) that you do...  Do you have any idea when we can expect to see the source and what license will it be released under?

Rem0o, 2020-03-10:
@MailYouLater I can see you hunted me down. Thanks for your interest in my personal project. 
Yes I've contributed to LHM to fix a few things. However, I don't have plans to open source FanControl and/or integrate it to LHM. "OpenFanControl" was the original name as I mixed "OpenHardwareMonitor" and "FanControl". It didn't mean I would be open sourcing it. Since then I renamed it to "FanControl" to avoid confusion but the original thread still keeps that name since I can't change it.

MailYouLater, 2020-03-10:
Not open source? ... So sad ...
😢
💔

Open source or not, your FanControl project is really cool and I hope to see it continue to improve.  I also want to say that I really appreciate your contributions to LibreHardwareMonitor.  Even if I don't happen to be one of the people who has hardware affected by your fix, every improvement or bugfix counts and I'm really happy that to see LibreHardwareMonitor continue to grow.  I hope to see more improvements and/or bugfixes from you for both LHM and FanControl.

Finally, if you ever stop developing FanControl (you know, like how SpeedFan died) I hope you'll consider releasing the code with an open source license instead of simply abandoning it, so that others can keep it alive like the LibreHardwareMonitor team is doing here with the OpenHardwareMonitor codebase.

Nama, 2020-06-25:
I made this. CLI interface, GUI is in the works. You can find pre-built exe files in Releases.
https://github.com/Nama/SuckControl/

lich426, 2020-07-13:
I have already created a fan control program that uses LibreHardwareMonitorLib.
[https://github.com/lich426/FanCtrl](https://github.com/lich426/FanCtrl)