LibreHardwareMonitor/LibreHardwareMonitor #172: Control the system fans ( not cpu ) with two curves

cednieb, 2019-11-30:
Hi

I search a free and open solution to control the speed pourcentage of  the system fan ( not the cpu or gpu fan ) .

I found a software ( Argus monitor ) but not free and not open source .

This software can adjust the speed of the fans with two curves ( speed fan / gpu temperature and speed fan /cpu temperature ) on the same time and it set the speed for the higher value found in the curves .

mb-asapp, 2020-05-05:
Hi! Which programs are you using or do you know, that can do this for CPU fan?

cednieb, 2020-05-06:
You can do this with the dll if you know c# or try use the trial of argus monitor ( https://www.argusmonitor.com/en/index.php ) .

xcasxcursex, 2020-05-08:
After speedfan became faulty two of my friends ended up searching for an alternative and there's practically nothing out there other than Argus. Apparently, it's very good and well worth the money, but for some of us it's not just about the free as in beer, it's more about the free as in freedom.

I'd LOVE to see this. I'd use it a ton. One problem I experience is that I can use my BIOS to control my case fans, but that's based on CPU (or motherboard/VRM) temperatures. That's fine if my CPU gets hot, but it doesn't account for my GPU. Being able to control the case fans based on both CPU and GPU temperature would be fantastic.