LibreHardwareMonitor/LibreHardwareMonitor #174: LibreHardwareMonitor doesn't support intel CometLake

frankie157, 2019-12-03:
In LibreHardwareMonitorLib/Hardware/Cpu/IntelCpu.cs, supported spec list doesn't contain intel Comet  Lake.

hirschmann, 2019-12-05:
Comet Lake, Amber Lake and Whiskey Lake are basically just Kaby Lake steppings aimed at mobile devices.
They share their (extended) model ID with Kaby Lake as well, therefore they are not separately listed in the MicroArchitecture enum in _IntelCpu.cs_, but they will be recognized by LHM.

See https://en.wikichip.org/wiki/intel/cpuid for more information.

If you have a Comet Lake CPU which LHM fails to recognize, please attach a report here (File --> Save report), so the issue can be investigated.

frankie157, 2019-12-06:
Thank you very much!
I close this issue.

closed 2019-12-06 by frankie157