LibreHardwareMonitor/LibreHardwareMonitor #175: idea for new functionality

mcoval, 2019-12-06:
Hi,
It would be great if you added to LHM 2 functionalities for remote web server:
- username of the currently logged in user,
- time from system boot.
LHM is a great application to monitoring multiple PC, but remote access doesn't show whitch user is currently logged on and how long he works on this machine. It would be very useful for some administrators.

cednieb, 2020-01-15:
Hi

For username : 
read this : https://stackoverflow.com/questions/1240373/how-do-i-get-the-current-username-in-net-using-c

System.Security.Principal.WindowsIdentity.GetCurrent().Name;  
or 
Environment.UserName
or
Request.LogonUserIdentity.Name;

Depends what you want .

For uptime
Read this : https://stackoverflow.com/questions/972105/retrieve-system-uptime-using-c-sharp
System.DateTime SystemStartTime = DateAndTime.Now.AddMilliseconds(-Environment.TickCount);
System.DateTime Uptime = DateAndTime.Now - SystemStartTime;

zamtmn, 2020-07-28:
Hi,
Please also add the current time display