LibreHardwareMonitor/LibreHardwareMonitor #180: LibreHardwareMonitor

bratskij, 2019-12-12:
Hey. I write with a translator, so that immediately apologize. Is there any Plugins for Rainmeter? OpenHardwareMonitor for this is, but it is not suitable for LibreHardwareMonitor. It is possible to do something new or edit an old one?
[OpenHardwareMonitorPlugin.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/3955634/OpenHardwareMonitorPlugin.zip)

radhaz, 2020-02-07:
I'm another person eager to use this in conjuction with Rainmeter. I have "Sidebar Diagnostics" for the time being but the way the data is presented in that tool isn't as customizable as Rainmeter.

black-ish, 2021-03-19:
Oh I would love to see that too.

nitehu, 2021-03-31:
Hello guys,
I've started a project to provide a rough LibreHardwareMonitor Plugin for Rainmeter. You may [check it out here](https://github.com/nitehu/Rainmeter-LibreHardwareMonitor).
Please let me know if it is working for you at all, or you have any problems with it.

closed 2021-09-28 by PhyxionNL