LibreHardwareMonitor/LibreHardwareMonitor #184: Gadget layout

TrickyDee42, 2019-12-16:
Is it possible to add, or add the option of adding a line between the different sections on the gadget.

i.e CPU info, then a gap before GPU info then a gap HD info.

Having all the info in one continuous list makes reading the info more difficult.

Just a thought.

Keep up the good work

TrickyDee42, 2020-01-03:
Or is this possible, and I've just missed it.

Love the gadget, and would like it to replace several others. But the compressed layout makes this difficult.

Another thought, is it possible to have multiple incidents of the gadget, could have a separate one for the different info, improving clarity.

Keep up the good work.

TrickyDee42, 2020-01-06:
Anyone else want to weigh in on this?

R.

TrickyDee42, 2020-01-09:
Is any of this possible @ArcadeRenegade

TrickyDee42, 2020-02-11:
Is it possible to control the colour of the lines of text shown on the gadget?

jdy4389, 2023-03-29:
I am wondering this as well, any possibility to re-arrange the displayed order of things on the gadget? Eg, to move CPU Temp above core clocks, or re-arrange the order of fans since Fan #&#x2060;1 doesn't always correspond with the CPU fan.