LibreHardwareMonitor/LibreHardwareMonitor #186: Incorrect temperatures for Radeon RX480

skibbipl, 2019-12-22:
I switched today from Open Hardware Monitor to Libre on my old computer and I found out that temperatures reported for my Radeon RX480 are incorrect:
[image]
HWiNFO reports them correctly:
[image]

fboldog, 2020-01-16:
I think it's easy to fix. Give me few days. ;)

Phyxion, 2020-01-29:
I have prevented these temperatures right now from showing with a simple check if it's within a legit range, but a better readout would be welcome @fboldog.