LibreHardwareMonitor/LibreHardwareMonitor #187: Error with .netcore project

ThaChillera, 2019-12-22:
Hi,

I'm converting the [sidebardiagnostics project](https://github.com/ThaChillera/SidebarDiagnostics) to .net core, and running into an issue. I changed the dependency to librehardwaremonitor's .net standard build, but it still fails due to a (seemingly) .net framework issue. Error report:

`{"Could not load file or assembly 'System.Management, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'. The system cannot find the file specified.":"System.Management, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a"}`

Stacktrace:
`"   at OpenHardwareMonitor.Hardware.SMBIOS..ctor()\r\n   at OpenHardwareMonitor.Hardware.Computer.Open()\r\n   at SidebarDiagnostics.Monitoring.MonitorManager..ctor(MonitorConfig[] config) in C:\\Users\\Robin\\Documents\\SidebarDiagnostics\\SidebarDiagnostics\\Monitoring.cs:line 34\r\n   at SidebarDiagnostics.Models.SidebarModel.InitMonitors() in C:\\Users\\Robin\\Documents\\SidebarDiagnostics\\SidebarDiagnostics\\SidebarModel.cs:line 100\r\n   at SidebarDiagnostics.Models.SidebarModel..ctor() in C:\\Users\\Robin\\Documents\\SidebarDiagnostics\\SidebarDiagnostics\\SidebarModel.cs:line 14\r\n   at SidebarDiagnostics.Sidebar.<<BindModel>b__10_0>d.MoveNext() in C:\\Users\\Robin\\Documents\\SidebarDiagnostics\\SidebarDiagnostics\\Sidebar.xaml.cs:line 180\r\n   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()\r\n   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)\r\n   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)\r\n   at System.Runtime.CompilerServices.TaskAwaiter.GetResult()\r\n   at SidebarDiagnostics.Sidebar.<BindModel>d__10.MoveNext() in C:\\Users\\Robin\\Documents\\SidebarDiagnostics\\SidebarDiagnostics\\Sidebar.xaml.cs:line 172\r\n   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()\r\n   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)\r\n   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)\r\n   at System.Runtime.CompilerServices.TaskAwaiter.GetResult()\r\n   at SidebarDiagnostics.Sidebar.<Initialize>d__7.MoveNext() in C:\\Users\\Robin\\Documents\\SidebarDiagnostics\\SidebarDiagnostics\\Sidebar.xaml.cs:line 107\r\n   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()\r\n   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)\r\n   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)\r\n   at System.Runtime.CompilerServices.TaskAwaiter.GetResult()\r\n   at SidebarDiagnostics.Sidebar.<Window_Loaded>d__19.MoveNext() in C:\\Users\\Robin\\Documents\\SidebarDiagnostics\\SidebarDiagnostics\\Sidebar.xaml.cs:line 245"`

Can anyone help me out why this error is happening?

closed 2020-01-17 by Phyxion

ThaChillera, 2020-01-30:
@Phyxion I'm still experiencing this error after pulling the latest version of LibreHardwareMonitor, and reloading nuget packages & doing a project clean & rebuild. For clarity, I'm using the netstandard2.0 build, not the net452 build.

Any idea why this would fail?

Phyxion, 2020-01-30:
Did you include System.Management from NuGet?

ThaChillera, 2020-01-30:
I did, I cleaned the project, cleaned nuget packages & rebuilt it.
That would be enough to download the latest version, correct?

Phyxion, 2020-01-31:
Hmm, it works fine for me in .NET Core (3.1). I just created a new .NET Core project to check again but it still works out of the box here. If you use a normal DLL reference you'll need to add 
`    <PackageReference Include="System.Management" Version="4.7.0" />` to your csproj (or use NuGet package manager). If you reference the project then System.Management will be automatically referenced by VS.

ThaChillera, 2020-02-01:
Thanks, this solved it, it was indeed a dependency issue.

ottovilles, 2020-07-24:
Hi @Phyxion!

This is an old issue, but I'm facing very similiar problem so I did not create a new issue. After adding a reference to System.Management, cleaning and building the project a new dependency problem appeared. This time error message was:

`System.IO.FileNotFoundException: 'Could not load file or assembly 'System.Threading.AccessControl, Version=4.0.5.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'. The system cannot find the file specified.'`

I installed a refrence to System.Threading.AccessControl via NuGet package manager, cleaned and build the project and tried to run it again. This error was resolved, but new error appeared:

`
System.MissingMethodException: 'Method not found: 'Void System.Threading.Mutex..ctor(Boolean, System.String, Boolean ByRef, System.Security.AccessControl.MutexSecurity)'.'
`

Any ideas how to solve this error?

commit b4739cee (2020-01-17): Fix #187