LibreHardwareMonitor/LibreHardwareMonitor #195: Asrock X570 Pro4 Support

Vargson, 2020-01-15:
The super io chip of the Asrock X570 Pro4 is apparently unknown.
It reports as id 0xD4 and revision 0x42.

Also, is there any way for me to add sort of "dummy" support for the chip? I really only need to access the fan controls.

Vargson, 2020-01-17:
update: currently working on a patch for it myself

Vargson, 2020-01-25:
update:
I have a partial patch for the board.
It uses the Nuvoton NCT6796D-R which is apparently very similar to the NCT6796D.

With the patch fan controls and sensors work. 
I am not sure how to check the voltages since the data points have different names than they do in other tools.
Temperature sensors are only partially working. It show one sensor that is unused and permanently stuck and the temperature on at least one other sensor needs an offset.

I don't really have the expertise to complete support for the board. Should I create a pull request regardless?

The patch is: 06182452cb85cd4e4355b7cd7018ca99fd195416

Phyxion, 2020-01-30:
Thanks for looking into this, I've added it in a2e6c8582e4a141f6a0457d44b32b2fa4f8ac82a. For setting the correct sensors, it's not that difficult. You need to add the motherboard in SuperIOHardware.cs in the correct section (`GetNuvotonConfigurationD`) and set the correct names based on what you can read in other software/BIOS. As a start, you can use the default implementation for the sensor. Sadly this is necessary for most motherboards because the motherboard manufacturers deviate from the standard sensor layout.

closed 2020-06-05 by Phyxion