LibreHardwareMonitor/LibreHardwareMonitor #196: GPU Package Power backwards for SLI

KiralyCraft, 2020-01-16:
I am running two 1080 Ti's in SLI on an Asus Prime x570 Pro motherboard. It looks to me that the power indicators for the GPU's are backward. When only one GPU is active driving monitors (all of them are plugged into the first, top-most GPU), it has the fastest clock speed of the two and should use the most power. However, according to LHM, they are switched.

[image]

Phyxion, 2020-01-31:
Thanks for the report, I suspect this happens because the ids used by NvApi are not the same as NvidiaML, I'll submit a change for this soon.

closed 2020-01-31 by Phyxion

KiralyCraft, 2020-01-31:
Yep, downloaded the new build, works perfectly now

commit d1cff075 (2020-01-31): Fixes #196

Switch to bus id instead of adapter index.