LibreHardwareMonitor/LibreHardwareMonitor #200: AMD CPUs temperatures incorrect

DeltaDesignRus, 2020-01-24:
We tested AMD processors, and the temperature of the ССD is not displayed correctly!
I am enclosing a set of screenshots. https://yadi.sk/d/5c2wS0sJpdgF2A
Write what you think about this, and is it worth waiting for a correction in the near future?

KiralyCraft, 2020-01-24:
It seems to work fine for the R7 3700x as well, since you didn't include it in the tests

DeltaDesignRus, 2020-01-24:
Yes, Zen2 temperatures are displayed correctly.

Phyxion, 2020-01-28:
I don't think CCD readings are available before Zen 2, I'll address this. I don't have an FX CPU available so I cannot verify that, so a PR for that would be appreciated.

DeltaDesignRus, 2020-02-01:
[image]
This video card does not display load

Ale008, 2020-02-07:
@DeltaDesignRus it does not display your GPU temperature neither. I have the same problem with my RX 580. Hope there's a fix soon.

Phyxion, 2020-02-11:
@fboldog Do you have any idea about this? I don't have an AMD card at the moment.

fboldog, 2020-02-11:
I will check, but it is a very busy week for me. 
I can't promise anything.