LibreHardwareMonitor/LibreHardwareMonitor #205: Command line version of HW monitor

totalnooob, 2020-02-11:
Hello is there a command line version of this software? I would need it to save log to txt file. Ive used different version of open hardware monitor with exe file OpenHardwareMonitorReport.exe but this show much higher cpu temperatures with threadripper cpus

Phyxion, 2020-02-11:
No, there isn't, you can easily write it yourself though by referencing LibreHardwareMonitorLib.

closed 2020-02-11 by Phyxion

totalnooob, 2020-02-11:
Can you do it ?

Phyxion, 2020-02-11:
You'll have to make it yourself, it's outside the scope of this.

totalnooob, 2020-02-11:
Im pretty bad at programming... Do i just compile this? Where can I change to use the libre dll?
https://github.com/openhardwaremonitor/openhardwaremonitor/pull/1000

totalnooob, 2020-02-11:
Ok nevermind found it