LibreHardwareMonitor/LibreHardwareMonitor #207: No FANs in the Lib

TheLaGmAn, 2020-02-15:
Im using the dll library but im getting no FANs report at all for any of the groups (cpu,gpu,motherboard).
I previously used OHM and it had an option to enable fan reports, but it doesnt appear on LHM dll, is that the issue? or FAN should be reported always just by setting IsCpuEnabled ?

cednieb, 2020-03-07:
computer.IsMotherboardEnabled = true; , I think .

TheLaGmAn, 2020-03-07:
its enabled, thats the issue, no fans even with that option

cednieb, 2020-03-07:
Try to launch as admin

TheLaGmAn, 2020-03-08:
I have it disabled right now but when i enabled that option, there was no fan as well, from what i understand that controller is for usb controllers and not for fans, is that right?

cednieb, 2020-03-08:
Launch as admin ?

The content of my folder for release x64 :

HidLibrary.dll
LibreHardwareMonitorLib.dll
LibreHardwareMonitorLib.sys
ListHotspotTemperatures.txt
LogitechLcd.dll
Monitor as admin.vbs
Monitor.exe
Monitor.exe.config
Monitor.pdb
MSIRGB.DLL.dll
MSIRGB.DLL.pdb

Msi and logitech is for my personnal use .

TheLaGmAn, 2020-03-08:
Yea, i have it as admin, i only use

LibreHardwareMonitorLib.dll
LibreHardwareMonitorLib.sys

since i dont need usb data, logitech or msi

TheLaGmAn, 2020-03-08:
i just tried again with iscontrollerenabled and no fans.
are you sure you get FANs? because that seems to be the only thing i dont get, and in OHM there was a specific option to enable those, it was "FanControllerEnabled" and in LHM thats not an option

TheLaGmAn, 2020-03-08:
this is the way i fetch the sensors

Dim GLOBAL_LHMD_Computer As LibreHardwareMonitor.Hardware.Computer
GLOBAL_LHMD_Computer = New LibreHardwareMonitor.Hardware.Computer()
GLOBAL_LHMD_Computer.Open()
GLOBAL_LHMD_Computer.IsCpuEnabled = True
GLOBAL_LHMD_Computer.IsGpuEnabled = True
GLOBAL_LHMD_Computer.IsControllerEnabled = True
GLOBAL_LHMD_Computer.IsMemoryEnabled = True
GLOBAL_LHMD_Computer.IsMotherboardEnabled = True
GLOBAL_LHMD_Computer.IsNetworkEnabled = True
GLOBAL_LHMD_Computer.IsStorageEnabled = True

For Each hw As LibreHardwareMonitor.Hardware.IHardware In GLOBAL_LHMD_Computer.Hardware
     hw.Update()

     For Each hws As LibreHardwareMonitor.Hardware.ISensor In hw.Sensors
     
          If hws.SensorType = LibreHardwareMonitor.Hardware.SensorType.Fan Then
                     MsgBox("FAN TEST")
           End If

      Next
Next

cednieb, 2020-03-08:
Ok the fans are in motherboard subhardware in the class

TheLaGmAn, 2020-03-08:
what do you mean?
hw.Subhardware is empty, 0 items, on each of the hws, hws.hardware.subhardware is also empty.

cednieb, 2020-03-09:
UpdateVisitor updateVisitor = new UpdateVisitor();
 
 Computer computer = new Computer();
          computer.Open();
          
          computer.IsCpuEnabled = true;
          computer.IsGpuEnabled = true;
          computer.IsMotherboardEnabled = true;
          computer.IsControllerEnabled = true;
          computer.IsMemoryEnabled = true;
          computer.IsNetworkEnabled = true;
          computer.IsStorageEnabled = true;    
          
          computer.Accept(updateVisitor);      
            
 for (int i = 0; i < computer.Hardware.Length; i++)
     {
     if (computer.Hardware[i].Sensors.Length > 0)
        {
        ...
        }
     
     if (computer.Hardware[i].SubHardware.Length > 0)
     for (int k = 0; k < computer.Hardware[i].SubHardware.Length; k++)
         {
         if (computer.Hardware[i].SubHardware[k].Sensors.Length > 0)
            {
            ...
            }
         }  
     }

cednieb, 2020-03-09:
`
 UpdateVisitor updateVisitor = new UpdateVisitor();
 
 Computer computer = new Computer();
          computer.Open();
          
          computer.IsCpuEnabled = true;
          computer.IsGpuEnabled = true;
          computer.IsMotherboardEnabled = true;
          computer.IsControllerEnabled = true;
          computer.IsMemoryEnabled = true;
          computer.IsNetworkEnabled = true;
          computer.IsStorageEnabled = true;    
          
          computer.Accept(updateVisitor);      
            
 for (int i = 0; i < computer.Hardware.Length; i++)
     {
     if (computer.Hardware[i].Sensors.Length > 0)
        {
        ...
        }
     
     if (computer.Hardware[i].SubHardware.Length > 0)
     for (int k = 0; k < computer.Hardware[i].SubHardware.Length; k++)
         {
         if (computer.Hardware[i].SubHardware[k].Sensors.Length > 0)
            {
            ...
            }
         }  
     }    
`

cednieb, 2020-03-09:
[image]
http://callfromother.space/unknown1.png

TheLaGmAn, 2020-03-09:
awesome, thanks a lot, i managed to get the motherboard sensors!

closed 2020-03-10 by Phyxion