LibreHardwareMonitor/LibreHardwareMonitor #208: No separation line between stacked plots

tnodet, 2020-02-23:
I noticed a minor GUI regression that appeared some revisions back.
The issue is that when "Stacked Axes" is checked and several x-axes (i.e. unit axes) are displayed in the plot panel, there is no separation (no black horizontal line) between the plots, where there used to be one.

The regression appeared in build 1.0.150 (commit [Migrate OxyPlot to NuGet version (#156)](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/bae89127b2d6c1813a9b8736a63a5c997541ff9a)).

Some screenshots will help clarify:

- Stacked axes in LibreHardawareMonitor 1.0.149:
[image]
- Stacked axes in LibreHardawareMonitor 1.0.173:
[image]

In the seccond screenshot, notice how the green curve of the top plot (GPU Core clock) is cut under a certain value, with no separation line.

lordmortis, 2020-04-05:
I've got a partial fix to this in a branch i'm working on. It puts the lines in, but it might double them up - but i've noticed another problem with plotting where it's not auto-setting the size of the plot properly and i want to take a look at that before putting in the PR. https://github.com/lordmortis/LibreHardwareMonitor/tree/fixedPlotting

closed 2020-04-07 by Phyxion

tnodet, 2020-04-09:
Thank you very much for the fix, @lordmortis :)
Can I ask, is there a particular reason you gave the lines a thickness of 2?
I just find them to be a bit too thick next to the outer plot borders, which are 1px thick.