LibreHardwareMonitor/LibreHardwareMonitor #211: Missing support Asus Prime B350M-A

dokotr, 2020-02-26:
Missing support of Asus Prime B350M-A, Super I/O: ITE IT8655E, not detecting temps, fans, voltage...

Phyxion, 2020-02-28:
Can you try the latest build (https://ci.appveyor.com/project/LibreHardwareMonitor/librehardwaremonitor)? IT8655E is included in there.

dokotr, 2020-02-29:
Thank you, just tried latest build 1.0.176 and no change, not readings from cpu or chasis fun, voltage or temp from mobo.
https://imgur.com/BZ8CfbK

Is there anything can do to help you to resolve it? Consider not programmer...

Phyxion, 2020-03-02:
Can you add the report here?

dokotr, 2020-03-06:
Sure
https://webshare.cz/#/file/2hs3182Os5/b350m-a-log

Phyxion, 2020-03-09:
Can you try the latest build? If it's still not working, can you also add the report of LHM?

dokotr, 2020-03-09:
Tried 1.0.178, not working but there is a change, ID of Super I/O chip shown up. Screenshot and report from LHM attached.

report:
https://webshare.cz/#/group/zQEtcGulYw/
Screenshot:
https://imgur.com/Kj785Ya

Phyxion, 2020-03-10:
Please try the latest build again.

dokotr, 2020-03-10:
Tried 1.0.179, same as on previous build, just showing name of super io. Report from LHM attached.

https://en.webshare.cz/#/group/oYOUdEWpvv/

dokotr, 2020-05-11:
Just tried latest build 1.0.190 and mobo sensor's are working ! Note sure whats gona change as I didn't find any description related to this issue in the change log.

closed 2020-05-11 by dokotr