LibreHardwareMonitor/LibreHardwareMonitor #212: Freeze and Crash when change resolution or switch displays

alpakeno, 2020-03-01:
Tried on Windows 10 x64 1803
It always do that, so I need to force closing it and reopen it.
I think it may need to be on background thread, instead of timer to prevent freeze and crash?

muddymind, 2020-03-05:
yeah, this bug is really annoying :\

Eno77, 2020-04-01:
Yes, I have this crash too.
Happens on my multi monitor setup sometimes by switching on/off the HDMI TV while other Monitors are in sleep mode.

Some additional Hint: The Samsung Magician up to version 6.0 suffered from the same bug and crashed at exactly the same time together with LHM. With actual version 6.1 it is fixed, only LHM crashes.

So maybe its related to SSD monitoring and monitor sleep mode / on/off switch / resolution changes causes the crash there?

I might do further tests, if somebody gives me advise / a version with crash log or something.

Eno77, 2020-04-21:
Update:
Samsung magician crashed again together with LHM. (So no "fix" and maybe related to SSD Monitoring)

closed 2020-05-25 by Phyxion