LibreHardwareMonitor/LibreHardwareMonitor #213: Expanding horizons

Soki324, 2020-03-02:
I want to build an external HW monitor and fan controller. I want to use libre hw monitor as a platform from which to read off as I do not know any C#. I also want to use your software so I can open-source my side so I don't have to update the entire thing every time a new MBO chipset comes out.

https://www.youtube.com/watch?v=iebBcQuBhYs

These guys had sort of a similar idea however there's no source and no way to customize anything. I want to use RPI connected with an MCU to do everything and make it open and customizable to at least some degree. 

I need from you to add options for webserver:

- Option to autostart with refresh enabled

- Option to select refresh window with that autostart

- Maybe like an HTTP "button" for external control of devices that are scraping the webserver, sort of like a website GPIO controllable from within the app (two of those if possible)

- An option to run the server on your local IP

I'm sure this will also make other people more interested in the development of such tools and solutions without fear of breaking their expensive PC.

cednieb, 2020-03-06:
Maybe add all the values in the registry create an application that would build a web page and distribute it on a web server component. In this way, the developers (users) could do their small projects

cednieb, 2020-03-06:
[image]
http://callfromother.space/unknown1.png

[image]
http://callfromother.space/unknown2.png

[image]
http://callfromother.space/unknown3.png

closed 2022-09-23 by PhyxionNL