LibreHardwareMonitor/LibreHardwareMonitor #220: ASRock X570 Phantom Gaming-ITX/TB3 Support

Mandosis, 2020-03-26:
According to HWiNFO64 the super io chip is the Nuvoton NCT6683D 

The LibreHardwareMonitor report states:
`Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xC732 at 0x2E/0x2F`

I am hoping to access the fan control

TheBengineer, 2020-05-02:
I am also hoping to control an ASRock x570 Phantom. If I get it working I'll let you know.

fabiensanglard, 2020-12-19:
Also unable to access states on my ASRock X570 Phantom Gaming-ITX/TB3

cloudcrypt, 2020-12-22:
+1 Same issue on this board

TheBengineer, 2020-12-22:
Fan control appears to be working on my ASRock X570 using [Fan Control](https://github.com/Rem0o/FanControl.Releases), which uses this library for hardware control.

iBeta666, 2021-01-14:
Having the same problem with the ASRock z490 Aqua.
Nuvoton NCT6683D/NCT5585D  are not correctly reported by Libre or OHM, but are by HWInfo64.

PsychoticWolf, 2021-03-16:
Still testing but I may have a version with support for the NCT6683D and this board... If it works I'll try and submit it.
[image]

mrsideshowjack, 2022-08-03:
Is this still open? 
I have the same issue but for the Z590 version `ASRock Z590 Phantom Gaming-ITX/TB4`, should I open a new issue?

[image]