LibreHardwareMonitor/LibreHardwareMonitor #224: Please sign your releases

buenalabs, 2020-04-03:
Windows UAC comes up every time I run the program.

Would it be possible to have signed releases?

For peace of mind and great security!

lordmortis, 2020-04-07:
In order to do this whomever does the releases would need to get a certificate (not accounting for possible changes in the build-chain, which I don't know anything about). These are, unfortunately, not free - not to mention there are security challenges with applying them. 

Not saying it's not possible, but there are a few barriers to this.

buenalabs, 2020-04-08:
You're totally right.

I was thinking about Certum and their free code signing certificates for open source developers. Just found out they're not free anymore.

Phyxion, 2020-05-03:
Sadly they are no longer free indeed, so unless someone is willing to donate a certificate, it likely won't happen.

the-j0k3r, 2020-05-08:
I see that the project has a prominent sponsor button, would this be a good way for interested people to contribute towards buying a code signing cert?

Perhaps add a donate tag to issues such as this that require funding to achieve.

Phyxion, 2020-05-25:
Sponsor button should be fixed now. The globalsign is only for website certificates, which doesn't work for signing executables.

ncaq, 2023-08-28:
Will UAC still be required if I register with Microsoft Store?
I heard that you can register a developer account here for a one-time fee of $20 and no renewal fee is required, so the burden will be much less.