LibreHardwareMonitor/LibreHardwareMonitor #229: Fan control on ITE IT8665E not working properly

nirvanaosc, 2020-04-21:
I have an Asus Crosshair VI Hero with IT8665E Super IO.

I am testing with a Be Quiet! Silent Wings 3 PWM 1500RPM fan connected to the system fan headers on the motherboard.

If I select "Manual" on the appropriate fan controller, the fan stays at 300 RPM despite which % I select. When I select "Default" the fan runs at 600 RPM.

javguit, 2020-04-24:
Changes to it87xx (and some it86xx) were made recently. Can you please update and test if works?

nirvanaosc, 2020-04-24:
Same results.

Rem0o, 2020-04-28:
Lookup over here... https://github.com/openhardwaremonitor/openhardwaremonitor/commit/7bf4f655d2c3ad9b660b6703c632b1d5694778ca

Phyxion, 2020-05-04:
[LHM.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/4573590/LHM.zip)

Can you give this a try?

pfto, 2020-05-04:
My issues are fixed with it. Everything working as expected, thank you.

commit 1104b661 (2020-05-05): Fix fan control for IT8xxx

Fixes #235
Fixes #229
Fixes #216
closed 2020-05-05 by Phyxion