LibreHardwareMonitor/LibreHardwareMonitor #235: Fan control on IT8792E/8795E not working & on IT8688E not working properly

pfto, 2020-04-27:
**Release:** 1.0.186
**Motherboard**: Gigabyte X570 Aorus Pro

*Problem with IT8688E:*

- Any value set for fan control within 0 to 50% spins fan up to around 65% RPM. 
- If set >50% the fan speed returns to current curve set in BIOS.

*Problem with IT8792E/8795E:*
- Any value set with fan control seems to be ignored and does not affect the fan speed.

Similar Issues: #229 #216

Phyxion, 2020-05-04:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/229#issuecomment-623334538

pfto, 2020-05-04:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/229#issuecomment-623531129

closed 2020-05-05 by Phyxion

rmattuschka, 2020-06-04:
The Size of the PWM Value in this register is only 7 bits [6:0]

You can try this in Row 258:	WriteByte(FAN_PWM_CTRL_EXT_REG[index], (byte)(value.Value >> 1));
instead of:      		WriteByte(FAN_PWM_CTRL_EXT_REG[index], value.Value);
                    
Maybe it works

CSchuetz7, 2020-06-06:
@rmattuschka 

Doesn't help either, unfortunately.

commit 1104b661 (2020-05-05): Fix fan control for IT8xxx

Fixes #235
Fixes #229
Fixes #216