LibreHardwareMonitor/LibreHardwareMonitor #236: Fans (RPM) not available via lib, SensorType.Control are showing up fine

uberhalit, 2020-04-28:
I can not access any Fan sensors (RPM) through the LibreHardwareMonitorLib, the control-sensors (speed-percentage) are working fine though. I am accessing the sensors like: 

```cs
Computer computer = new Computer();
computer.IsControllerEnabled = true;
computer.IsMotherboardEnabled = true;
computer.Open();
List<ISensor> sensors = new List<ISensor>();
foreach (IHardware computerHardware in computer.Hardware)
{
    computerHardware.Update();
    foreach (ISensor sensor in computerHardware.Sensors)
    {
        sensor.Hardware.Update();
        sensors.Add(sensor);
    }
    foreach (IHardware sensorController in computerHardware.SubHardware)
    {
        foreach (ISensor sensor in sensorController.Sensors)
        {
            sensor.Hardware.Update();
            sensors.Add(sensor);
        }
    }
}
```
[image]

As you can see there are 6 `SensorType.Control` but no `SensorType.Fan`. This is how it looks like in the utility: 
[image]

There are **7** sensors, both Control type and Fan type. Fan 2 and 5 are reporting 0 RPM though.

riQQ, 2020-04-28:
Have you checked #207?

uberhalit, 2020-04-28:
> 
> 
> Have you checked #207?

Yes, it's not the same issue. #207 didn't enumerate SubHardware, my problem is Fan sensors are not available at all.

Phyxion, 2020-05-04:
You need to recursively update the hardware, and run as admin. Usually the fans are available only on the second update as they are activated after the first update.

closed 2020-05-04 by Phyxion