LibreHardwareMonitor/LibreHardwareMonitor #237: Bogus CCD sensors dynamically added with incorrect values

aaronjwood, 2020-04-30:
A picture is worth a thousand words :)
[image]

All I did was leave LHW open for some time as I was compiling some stuff and saw this behavior in order:

1. Max and average CCD temp values went off the charts
2. Bogus CCD sensors popped up shortly after with the same crazy values

chitza, 2020-04-30:
Is that a Ryzen CPU? I remember seeing something like that too, but I can't reproduce it anymore.

aaronjwood, 2020-04-30:
Yeah, sorry I should have specified that I'm running a 3950x on X570 (ASRock Taichi). I'm not sure how to reproduce it either, so far I've only seen it once in a fairly heavy-load situation. This was also with the latest release zip from a few hours ago too.

Phyxion, 2020-05-04:
Thanks for the report, this should no longer occur. If you still see it, please let me know.

commit ec01af6c (2020-05-04): Update Amd17Cpu.cs

Fixes #237
closed 2020-05-04 by Phyxion