LibreHardwareMonitor/LibreHardwareMonitor #238: Feature request: Hide node from tree view

lbreuss, 2020-05-01:
I would like to hide some unneeded nodes from the tree view. 

For example, I would like to see the temperatures of all devices, but I don't care about voltage, HDD Load/Usage/Throughput. Each HDD is still present with these 3 collapsed nodes taking up visual space.

chitza, 2020-05-01:
FYI, I've recently implemented this in #222
The latest build should contain the feature.

lbreuss, 2020-05-02:
Great, thanks. Tested the latest build. As a bonus, the tree node vanishes from the view when all leafs have been hidden.

How do I hide all 16 Core loads at once? A context menu action "Hide" on the "Load" node, and on the top level device would be nice.
How do I unhide an accidentally hidden leaf value, e.g. the aggregated Load?
How do I unhide a vanished intermediate node? An unhide on Load and on the CPU node would be nice. I would prefer a GUI feature, rather than manually editing the config file, or doing a full reset of the view.

chitza, 2020-05-04:
I think I've might of misread your initial comment. What I've implemented is node state persistence - i.e.: if you close a node, it will remain closed unpon subsequent restarts of the app.
I didn't work on node hiding at all.