LibreHardwareMonitor/LibreHardwareMonitor #240: openhardwaremonitor got update this year 0.9.x will you merge it somehow?

informatorius, 2020-05-04:
openhardwaremonitor got update this year 0.9.x will you merge it somehow?

xcasxcursex, 2020-05-04:
I was about to post the same. There are already a lot of competing applications in the closed-source space, so it would be a shame to see an open-source alternative competing with its fork.

Phyxion, 2020-05-05:
I've contacted the OHM author a while back but he has zero interest to continue with LHM code (which has a lot more features and is greatly cleaned up), so it won't happen. Not my choice, but you can complain at OHM about it.

closed 2020-05-05 by Phyxion

informatorius, 2020-05-05:
Does it make sense to take over some code or hw specs of openhardwaremonitor 0.9.x or is LibreHardwareMonitor already more advanced?

xcasxcursex, 2020-05-05:
Pulling code from 'upstream' would be good in the meantime.