LibreHardwareMonitor/LibreHardwareMonitor #246: Support to IT8655E

TheTrueBumba, 2020-05-12:
Worked until latest release and stopped now. I will try to check the code and find a solution. Can you check in your side?
Thanks in advance.
[image]

Phyxion, 2020-05-13:
Please take a look, I don't have any IT devices here to try it on.

TheTrueBumba, 2020-05-13:
I checked the code. IT8655E enters in "generics sensors" code. I will try to identify how check fan sensors in my motherboard and paste the code here.
Open Hardware Monitor doesnt detect my sensor, BTW.

Phyxion, 2020-05-14:
Weird, please let me know if you find the correct ID for it.

TheTrueBumba, 2020-05-14:
I checked better. If I understood your code correctly, it return empty em fan sensors collection. Some paid solutions capture my ASUS Prime B450 correctly. I saw a library using WMI to capture the sensors. The library was developed by a third part developer? I can atach here the debug screen from my Visual Studio. Fan collection (The others sensors too) empty before choosing the motherboard manufacturer.
Very, very weird.
I am learning your code and I will install Visual Studio in another machine to compare the sensors detection using a older sensor and motherboard model.

Phyxion, 2020-05-25:
You can check in LpcIO and IT87XX where things go wrong, I don't have a device with this sensor.

TheTrueBumba, 2020-05-30:
Tnxs. I found the issue, now I need to know how to resolve or help you to resolve it.

In LPCIO.cs, line 174, you have the following code:
superIOs.Add(new IT87XX(chip, address, gpioAddress, version));

In IT87XX.cs, line 187, you have the condition below:
if (!valid || vendorId != ITE_VENDOR_ID)
  return;

In my motherboard, Asus PRIME B450M-GAMING/BR, vendorId has a value of 127 and ITE_VENDOR_ID has a value of 144, so the code step out the sensor chec. If I jump to line 191 and just hit "F5", everything runs fine since the code detects my IE correctly:

case Chip.IT8655E:
voltages = new float?[9];
temperatures = new float?[6];
fans = new float?[3];
break;

The question: Why vendorId and ITE_VENDOR_ID has different values? A not reconized motherboard? A issue? You guide me I can download the latest version, make the corretions and make a push.

I will keep studying here. Tnxs for the advice.

TheTrueBumba, 2020-05-30:
Well, working here, but I dont know the reason since I am not a specialist in hardware sensors.
But changing the const ITE_VENDOR_ID to 0x7F worked like a charm.
private const byte ITE_VENDOR_ID = 0x7F; (Line 40 in IT87XX.cs)
Are there any sensor database so I can check the correct vendor ID?

TheTrueBumba, 2020-05-30:
[image]

Phyxion, 2020-06-02:
Thanks for checking, I've added the other vendor id as well.

closed 2020-06-02 by Phyxion

oxirt, 2020-11-08:
Please note that the fan readings are incorrect. See ticket 277.

commit 5976a5a7 (2020-06-02): Update IT87XX.cs

Fixes #265
Fixes #246