LibreHardwareMonitor/LibreHardwareMonitor #251: Problem with ITE IT8792E/IT8795E - values are ignored

Shasoosh, 2020-05-20:
Release: 1.0.191
Motherboard: Gigabyte Z390 Aorus Ultra

Problem with IT8792E/8795E:

Any value set with fan control seems to be ignored and does not affect the fan speed.

I see that a similar issue was addressed in [235](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/235) but unfortunately I'm still experiencing it.

CSchuetz7, 2020-06-04:
Same Problem for me as well.

[image]

Phyxion, 2020-06-05:
Please check the latest build, this might be fixed now.

Shasoosh, 2020-06-05:
Still not fixed for me.

CSchuetz7, 2020-06-05:
Problem still exists with the latest version.

Phyxion, 2020-06-05:
Does latest OHM work for you?

CSchuetz7, 2020-06-05:
I have the same problem with latest OHM.

Shasoosh, 2020-08-01:
Any updates on this?

Phyxion, 2020-08-02:
We'd need a developer with this hardware first.

CreeT6, 2021-01-24:
is adding support for this chip easy? can one learn how to do it?

PhyxionNL, 2021-01-25:
If you know how to code, have a motherboard with it to test, and preferably get the documentation for this chip, then yes.

NineBallAYAYA, 2021-02-23:
@PhyxionNL (sorry for the ping didnt know who else to ask)
**Short and sweet**
- Are there any risks if I do something wrong? 
- Where should I look for documentation if google has none?

**Ramble**
I have a mobo with the ITE IT8792E/IT8795E and although this stuff is is out of my normal comfort zone I have an excess of free time to learn. I would be willing to bang my head into a keyboard for a few hours but I cant find any documentation on the chip at all, what I have found is a really old linux driver I might be able to make use of. This might be a noobie question but I have little experiance/Questionable documentation, is there a risk of frying my superio/fans if I mapped the wrong values?

PhyxionNL, 2021-02-23:
@NineBallAYAYA I don't think so. The best source of documentation is the Linux driver at the moment, https://github.com/torvalds/linux/blob/master/drivers/hwmon/it87.c
But it remains a bit of trial and error.

NineBallAYAYA, 2021-02-23:
@PhyxionNL Thanks a lot, was quite worried i would slip up and brick something. Ill give it a shot tommorow and hopefully i can get something working somewhat soon.

PhyxionNL, 2021-02-23:
It's probably a small thing that needs changing, like a different registry or that it uses ext reg. You can experiment a bit with the possibilities here: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/7ac7c29013c6292a43d14d4da34ef5845b11a13b/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/IT87XX.cs and base it off the Linux driver where possible (you can compare the features with other IT sensors).

jeffdamp-wave, 2021-03-03:
has anyone made any progress on the IT8792E chip? I just cloned the enlistment and added the controls in but need to figure out the correct reg values.  

I'm on an Auros Master X570.  I'm able to control on fans on the IT8688E chip.

jeffdamp-wave, 2021-03-03:
I took a look at the it87.c file above and the code in the lib.  They seem to be in sync with each other (other than the linux file using 6 byte arrays which I added to the lib.  No luck making those changes either.  I wonder if fan control 5 by gigabytes has the values and could be decompiled.

NineBallAYAYA, 2021-03-03:
> has anyone made any progress on the IT8792E chip? I just cloned the enlistment and added the controls in but need to figure out the correct reg values.
> 
> I'm on an Auros Master X570. I'm able to control on fans on the IT8688E chip.

Not much on my attempts.. Most ive figured out is that someone partially implemented the codepath for it under the name "IT879XE (DEVID = 0x8733)" this explains why it can read values. I poked around a lot and most ive figured is its *probably* just writing wrong reg values / addresses. So I basically came to the same conclusion as you but I havent figured out a solution yet so keep me posted if you figure something out.

CreeT6, 2021-03-03:
> I took a look at the it87.c file above and the code in the lib. They seem to be in sync with each other (other than the linux file using 6 byte arrays which I added to the lib. No luck making those changes either. I wonder if fan control 5 by gigabytes has the values and could be decompiled.

can you use fancontrol5 to control those fans while checking which reg values change?

jeffdamp-wave, 2021-03-04:
> can you use fancontrol5 to control those fans while checking which reg values change?
I don't know how you would see this since these are the addressable ports sent to the driver.

NineBallAYAYA, 2021-03-04:
> can you use fancontrol5 to control those fans while checking which reg values change?

This was my original plan but as jeff said, i dont think it would be possible unless you manged to get between fancontrol and the chip and i dont know how you would do that on windows.

However it might be possible to modify the linux driver so it logs the values sent to the chip and that should* be fairly easy. As a last resort just opening control 5 in cutter/ida/ghidra might work but will probably take forever to find anything of use.

jeffdamp-wave, 2021-03-04:
I took a second look at the issue on my Gigabyte X570 this morning.  The register values are correctly addressing the fan speeds.  The problem is that the BIOS is maintaining control over the fans.  By spinning in a loop on the call to SetControl I could see the fans actually spinning up and down but only for the length of time I was constantly setting the value.
[image].  

I wonder if there is a secondary command that needs to get sent to let the BIOS know the fan is controlled in software.

CreeT6, 2021-03-04:
> I took a second look at the issue on my Gigabyte X570 this morning. The register values are correctly addressing the fan speeds. The problem is that the BIOS is maintaining control over the fans. By [...]

Now that i know someone else is also having the same strange issues, I can also confirm this on my x570 pro. On my AIO pump, bios is set to 50%. When I set the speed to 0%, the pump dips for a second then back to 50%. 1350rpm steady, then dips to 1200 rpm then back to 1350rpm, very audible for my pump.

I guess, whatever the code is doing to the IT8688E (other chip on MB that respond correctly to fan control) is not working for IT8792E

PhyxionNL, 2021-03-04:
Probably the values written in SetControl are incorrect, one of them should specify manual control. It might be related to the old_peci value in it87 driver.

Shasoosh, 2021-07-30:
Hey guys, how can I help debug this?

Shasoosh, 2021-08-04:
Adding some info to this:

Found a pdf with the IT8792E addresses here: https://github.com/richard378/it87-eol

I've also found this commit which indicates that there is some kind of problem leaving the configuration mode of the chip. Not sure how much it's relevant but I'll add it here for future reference:

https://github.com/richard378/it87-eol/commit/485178c1ecc1835066178eeda681a4d0a9cdd0ef

I hope this helps.

palvarad, 2021-10-05:
Hello and thank you for creating such a cool tool,

I have a Z370 Aorus Gaming 7 with the ITE IT8792E chip and recently started using FanControl to replace Gigabyte's SIV.
I really like FanControl, but I am running into the same issue that I can only control half my fans.

Based on issue 69 in the FanControl Git and this one it seems the issue has been narrowed down to having to set the fan to manual control.
I have a little programming experience and would like to help. I was able to download and compile LibreHardwareMonitor.
I also made a few changes to try and test different values. However, I need a little guidance on how/what to test.

Below is what I have done so far, please let me know if anything needs to change.

1 - Added **IT8792E** to the **Chip.cs** and assigned the value of **0x8732** (I saw this number in the PDF provided by Shasoosh).
2 - In IT87XX.cs I added **chip == Chip.IT8792E** to **_hasExtReg** and added a new case matching the values of case **Chip.IT879XE**
3 - In IT87XX.cs inside SetControl I added **if(Chip == Chip.IT8792E)** inside **if (_hasExtReg)**. 

Inside the new if is where I think we can test different values and should only apply to IT8792E.
For testing, should I copy the lines below inside the new if and test different values by replacing 0x7F?

WriteByte(FAN_PWM_CTRL_REG[index], (byte)(_initialFanPwmControl[index] & 0x7F));
WriteByte(FAN_PWM_CTRL_EXT_REG[index], value.Value);

Thanks again

maker-DJTodd, 2022-04-27:
Experiencing same issue with a system on gigabyte board. Z270x gaming 7.

BenNeumeister, 2022-04-29:
Hi all,
Looking into this topic in my x570 Ultra and found this thread.

Issue appears to be that the 879x chips use Ctrl Address 0x13 and 0x14 differently to the 'other' chips.
For 879x you need to use 0x14 and 0x13 is inverse to the others.
PWM granularity is also not the same either.
I cant set the PWM locally in my system but its not working as expected yet.

I'll attempt to sort this out over the next few days, but atleast I can change the RPM.

Koli0842, 2022-05-15:
Has there been any movement on this one? I won't say I'm the best person to seek for help, but I have one of the affected motherboards, if some testing needs to be done (X570 Pro)

Elektrobomb, 2022-06-04:
Just leaving my bit in here as well, Aorus AX370 Gaming 5 has the same issue with the IT8792E chip onboard.  Fan control not working correctly with the exact same symptoms as above.  I do know embedded firmware stuff so I'll see if I can have a gander but I've never done hardware level stuff on windows so it's probably gonna be a bust

Digital-Storm, 2022-06-18:
Can this be reverse engineered from Gigabyte's SIV software? That software seems to have the ability to change the fan values.

Digital-Storm, 2022-06-18:
There seems to be a method that uses WMI, perhaps this can be adapted and implemented here as well.
https://github.com/t-8ch/linux-gigabyte-wmi-driver

Keltere, 2022-06-18:
> There seems to be a method that uses WMI, perhaps this can be adapted and implemented here as well. https://github.com/t-8ch/linux-gigabyte-wmi-driver

Not compatible at least with aorus master https://github.com/t-8ch/linux-gigabyte-wmi-driver/issues/7

javs, 2022-11-13:
It appears the 2nd chip in those MB is handled by something else. I don't think it's being driven wrong by the code here, it's just that there is another component/firmware actively driving it in the bios, so the values get overwritten. That's why the experiments here are not succeeding with these MBs.

There is another memory area that has the controls for driving this, which is very likely specific to Gigabyte MBs. I guess something could be written detect and drive that instead of the ITE chip directly, that would certainly make this work.

I've tested with a Gigabyte AORUS B550 Pro, it has a ITE86xx driving the first 5 fans (which works fine), and this ITE8795E as a second chip driving the last 3 fans.

Something that I noticed that helps confirm this idea, is that the 2nd chip, in SIV (GB's program for fans), supports using the VGA as a temperature input (i.e. not the sensor they put under the PCI16X slot, but the actual GPU reading), where the 1st chip doesn't. Even when stopping the EasyTune service, the temp reading is still driving the fan correctly, so this points to that other component/firmware not only driving the 2nd chip, but also handling this other interaction with the GPU.

LightTreasure, 2022-11-18:
Going deep into multiple rabbit holes, it seems like there might be a fork of the it87 driver that *could* have the fix for this issue.

It's maintained by Frank Crawford : https://github.com/frankcrawford/it87 

Specifically, check the following thread on the unraid forum: https://forums.unraid.net/topic/92865-support-ich777-amd-vendor-reset-coraltpu-hpsahba/page/52/

Also the following code comment in it87.c from Frank's fork sheds some light:

`/*
 * On some Gigabyte boards sensors are marked as ACPI regions but not
 * really handled by ACPI calls, as they return no data.
 * Most commonly this is seen on boards with multiple ITE chips.
 * In this case we just ignore the failure and continue on.
 * This is effectively the same as the use of either
 * acpi_enforce_resources=lax (kernel)
 * or
 * ignore_resource_conflict=1 (it87)
 * but set programatically.
 */`

Koli0842, 2023-03-09:
I am still aching for this to work, so I fired up a dev environment with zero experience in this field. I was able to confirm that:
- Indeed the last [0-2] bits of 0x14 are for turning on/off the fans, does not seem to affect anything
- [3] bit there seems to be 0, not sure what it means
- [4-6] should be pwm frequency, 6 is 1, others are 0 for us
- Last bit [8] there seem to be polarity at 1, that one seems correct for us.
- Last [0-2] of 0x13 are control, normal code path sets this 1 before overriding the value, does not seem to work for us either way
- [3] bit seems to be 0, not sure what it means
- [4-6] should be tachometer. By default they are 1, when 0 no tachs are reported
- [7] bit seems to be 0, not sure what it means

- Sometimes it indeed lets you control it for a moment before resetting. From my experience, setting a lower than the last value makes it switch momentarily before reverting, but selecting the same or higher value again makes no difference.

javs, 2023-03-09:
I have test branch that can successfully drive the 2nd IT879XE in my Gigabyte B550 Aorus Pro, see https://github.com/javs/LibreHardwareMonitor/tree/issue251

A few important caveats: this only works for Gigabyte AMD boards at the moment (of which I only tested mine), and it won't perform any kind of discovery, so it may mess up your stuff if what it wants is not present on your hardware.

What it does is temporarily turn on MMIO for a block of memory coming from LPC via the AMD ISA bridge, that holds a few more registers related to fan control, which seem to be Gigabyte specific. These registers act as an override for the controls of the 2nd ITE chip in the board, so as soon as they were changed with LibreHardwareMonitor, the GB controller was resetting them. Offset 47h in it has a master on/off, which once disabled, makes the current unmodified IT87XX code start working normally.

I suspect this GB controller self disables if it doesn't detect a windows boot, or maybe the GB kernel driver is enabling it when it loads. This may explain why the it87 linux driver works fine even though it has similar driving code than LibreHardwareMonitor. I did try to disable all GB services and boot but the controller is still enabled somwhoe, but maybe there is something to this someone could explore.

As it stands now, there are two missing things before anyone can send a PR of this: we would need discovery code to understand when is it safe to execute this code, and also we need an Intel variant most likely, which could be written.

Koli0842, 2023-03-09:
@javs
I can confirm it is working in the X570 Aorus Pro aswell. Man, you're bloody awesome, what a difference it makes when you know what you're doing.. :D

I have a friend with an Intel board having the sam issue, I assume they might have different memory addresses for their ISA bridge?

For identificaion I assume we could just pass the motherboard models to LpcIO.cs from Motherboard.cs's constructor, as it's identified already at that point. Not sure it's the most sophisticated approach.

javs, 2023-03-09:
@PhyxionNL I'm interested in your thoughts here about me making a PR about this approach (see the last couple of comments).

Would we whitelist this based on motherboards? or try and write some discovery code based on what the GB tools do?

Also, I presume you'd want this to only run when the user tries to change control of a fan, unlike what I have now that blindly disables it.

Koli0842, 2023-03-10:
@javs It would make sense to move this to the control bit, when trying to use this within https://github.com/Rem0o/FanControl.Releases, it does not touch the current codepath. After wakes from sleeps, I need to start LibreHardwareMonitor first, then re-enable control of the fans from FanControl.
How exactly did you find the required addresses? Is it from SIV, or the linked AMD ISA PDF itself? I have access to a friend's Z390 system if something needs to be looked at.

PhyxionNL, 2023-03-10:
> @PhyxionNL I'm interested in your thoughts here about me making a PR about this approach (see the last couple of comments).
>
> Would we whitelist this based on motherboards? or try and write some [...]

I think we can do some manufacturer specific things in LpcIO to address this. So pass through the motherboard model to LpcIO and then check and disable accordingly. The ASUS Embedded Controller also passes in the model (check line 73 of Motherboard.cs).

javs, 2023-03-10:
@Koli0842 I see now the same behavior, both fan control not hitting that and the resetting on sleep/resume, I'll check it out.

I got the values and steps by observing what SIV does. The process seems similar for Intel with a few additions I'm not sure about yet, and also different offsets, registers and addresses.  What the code is doing is map the bios flash to ram and make changes there. There is a brief explanation of the process here: https://stackoverflow.com/questions/60200796/how-does-processor-read-bios-from-spi-flash (only I think GB uses LPC instead of SPI, different bits to configure). I can get a rough approximation and we could go from there. I can also give you some pointers offline if you are interested.

@PhyxionNL I'll take a look, thanks!

javs, 2023-03-11:
Sent a version to my branch that has the suggested changes https://github.com/javs/LibreHardwareMonitor/tree/issue251. I would like to get rid of the sleep somehow and also find a better solution for the sleep problem to avoid forcing the state on every fan control set.

However, it still doesn't work when used in FanControl. Looking a bit into it, it seems the 2nd kernel driver that we use here, InpOut, doesnt work when loaded in FanControl. This [ticket](https://github.com/Rem0o/FanControl.Releases/issues/279) makes me suspicious it was manually disabled there, as InpOut.IsOpen is true the first time we enter Open() in FanControl when it shouldn't. CC @Rem0o

Unfortunately without that driver I don't see an alternative to drive this controller. Maybe after [984](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/984) is done we can use Ring0 instead?

Rem0o, 2023-03-11:
@javs  yep I manually disable InpOut using reflection. Caused various issues at some point and it wasn't used much in any way that benefited FanControl.

javs, 2023-03-11:
@Rem0o I noticed you are involved in https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/984, maybe if you are close there, and can enable `_PHYSICAL_MEMORY_SUPPORT ` I could rewrite this code to not use InpOut.

For others, I've added a copy of the fix as a standalone program you can run manually here https://github.com/javs/GigabyteFanController. Note this is AMD only, still no checks performed, be careful. Run the program as admin with `false` as argument, then use LibreHardwareMonitor / FanControl as usual and it should work (set FanControl to force apply one of the fans). Needs to be rerun on resume, you can do that with a windows scheduled task.

Rem0o, 2023-03-11:
@javs  Sadly we are at at stand-still on that issue, since nobody in here has an EV code signing cert, since being incorporated is required.

Koli0842, 2023-03-12:
> I can also give you some pointers offline if you are interested.

Sincerely thanks for the offer! In all honesty, while I find the area fascinating and I don't mind getting my hands dirty, it's so far from my field of expertise, I see myself more of a burden than help. :D
That being said, I am available for things I can potentially help with, including talking to my friend with the Z390 board to see how things are on the Intel side.

PhyxionNL, 2023-03-13:
> Would it make sense to contact Gigabyte about this issue and open a request there?
> 
> I just stumbled upon this fork: https://github.com/frankcrawford/it87

This also does some specific Gigabyte stuff, might be worth checking this @javs.

javs, 2023-03-19:
I've added better discovery code under https://github.com/javs/LibreHardwareMonitor/tree/issue251 that should work with more AMD based GB boards. I'll try it for a few days and send a PR.

BenNeumeister, 2023-03-19:
> I've added better discovery code under https://github.com/javs/LibreHardwareMonitor/tree/issue251 that should work with more AMD based GB boards. I'll try it for a few days and send a PR.

Nice work!
Can confirm it works on my x570 Ultra.
Needed to add the fan controls to the code but after that but loads the current values and then also can control them via the frontEnd for LibreHWM.
I added the .dll to FunControl, it can see them all but cant control the fans 4->6 fans from there for some reason....

javs, 2023-03-19:
@BenNeumeister nice, thanks. FanControl won't work for the moment (see above [comment](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/251#issuecomment-1464969469)).

Could you please retest with the code I pushed just now (same [branch](https://github.com/javs/LibreHardwareMonitor/tree/issue251)). If you could also try the following test it would be useful as well:
1. Set service EasyTuneEngineService to disabled.
2. Set scheduled tasks and SIV and SIV-VGA to disabled.
3. Restart PC. Note this will disable your SIV curves and set your fan configs to whatever your bios has.
4. Try controlling fans in LHM.
5. Tell us what are the models of the two ITE controllers your MB has.
6. Undo 1 and 2 and start services as needed.

BenNeumeister, 2023-03-20:
> 

> @BenNeumeister nice, thanks. FanControl won't work for the moment (see above [comment](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/251#issuecomment-1464969469)).
>
> Could you [...]

With you update I just had to add this to SuperIOHardware.cs;

c.Add(new Ctrl("Fan Control #5 Pump", 0));
c.Add(new Ctrl("Fan Control #4", 1));
c.Add(new Ctrl("Fan Control #6 Pump", 2));

To line 1769.

First Controller is an ITE IT8688E
2nd is an ITE IT8791E/IT8792E

I can control all devices I've plugged in from 0% to 100%.
System Fan #2 (IT8688E)
Fan Control #5 Pump/ Fan Control #4/Fan Control #6 Pump (ITE IT8791E/IT8792E)

closed 2023-03-21 by PhyxionNL

javs, 2023-04-06:
For the Gigabyte Intel gang watching this, I'm taking a stab at trying to write something to make this work.

Could anyone with one of the non-working Intel boards that can debug the latest LHM tell me what is the value of the `address` parameter when `GigabyteController()` (in GigabyteController.cs) is called?

citrixscu, 2023-04-07:
Thanks for your work on this @javs . Quick question -- is the fix valid for `IT87952E` as seen on Gigabyte Aorus X570S? I looked in SuperIOHardware.cs and do not see a corresponding `case Chip.IT87952E:`; I cannot control the fan(s) on this chip using your [https://github.com/javs/GigabyteFanController](https://github.com/javs/GigabyteFanController) so I was curious.

javs, 2023-04-07:
> Thanks for your work on this @javs . Quick question -- is the fix valid for `IT87952E` as seen on Gigabyte Aorus X570S? I looked in SuperIOHardware.cs and do not see a corresponding `case [...]

A few thoughts to see if I get you:
1. When you say SuperIOHardware.cs and IT879**52**E did you mean LpcIO.cs `FindGigabyteEC()` and IT879**5**E ?
2. Is your MB listed in SuperIOHardware.cs and are all fan controls showing?
3. Does https://github.com/javs/GigabyteFanController show any error?

citrixscu, 2023-04-07:
> A few thoughts to see if I get you:

> 1. When you say SuperIOHardware.cs and IT879**52**E did you mean LpcIO.cs `FindGigabyteEC()` and IT879**5**E ?

I was looking only in SuperIOHardware.cs.

> 2. Is your MB listed in SuperIOHardware.cs and are all fan controls showing when you run LHM?

Not explicitly. I have the X570**S** AORUS MASTER, whereas SuperIOHardware.cs has an entry for `Model.X570_AORUS_MASTER`. Fan controls show and work correctly for anything under the ITE IT8689E chip. LHM identifies the board's ITE IT8795**2**E as an ITE IT8795E; RPM is accurate under this chip but the controls %s are incorrect and do not control any fan(s) connected.

LHM 0.92 Output:
[image]

HWiNFO64 v7.42 Output:
[image]

> 3. Does https://github.com/javs/GigabyteFanController output any error? 

No errors thrown when using the application. I can successfully change between `true` and `false` and get the expected output, but there is no change in being able to control via LHM or FanControl.

javs, 2023-04-08:
Ah, yes. See below, that chip needs further work as the registers that identify the memory mapping are different.
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/0196dca04a6e73eaa55b522b7496fd0722ac6fbd/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/LpcIO.cs#L640-L646

Regarding the name, HwInfo and the it87 linux driver identify your chip as IT87952E, which has one extra number than all others. However, Gigabyte software identifies that chip as IT8795E. There doesn't seem to be specs for this, and screenshots online show it as 8795 anyway, so in LHM I went with that.

vincesmits, 2023-05-08:
Hey @javs i have a System Board Model Z590 AORUS PRO AX (INTEL) Can i provide you anything ?
did you get a hold on _"Could anyone with one of the non-working Intel boards that can debug the latest LHM tell me what is the value of the address parameter when GigabyteController() (in GigabyteController.cs) is called?"_

javs, 2023-05-09:
> Hey @javs i have a System Board Model Z590 AORUS PRO AX (INTEL) Can i provide you anything ?
>
> did you get a hold on _"Could anyone with one of the non-working Intel boards that can debug the latest [...]

Yeah, i'm still looking for that value to confirm i'm on the right track.

mrehkopf, 2023-05-18:
@javs 
> Ah, yes. See below, that chip needs further work as the registers that identify the memory mapping are different.

Same problem here with an X570S AORUS MASTER. I tried your code from the GigabyteFanController repository as well. The problem is that the EC doesn't seem to have MMIO mapping of SRAM enabled; your memory mapping code only returns a block of all FF's. I tested various other regions and MMIO mapping *in general* seems to work; just the "IT87952" doesn't return any data with the firmware it's running.

For comparison I ran HWiNFO64 in debug mode and it was able logged an EC RAM dump with the relevant block (I *think* - please confirm!) at 0x10900 instead 0x900:

```
10900: 01 00 1E 2D 3D 1E 1E 20 35 C9 2E 00 00 00 00 00
10910: 00 74 FF C0 22 00 51 08 00 00 00 00 00 00 00 00
10920: 00 00 00 00 00 00 00 00 00 00 00 00 9E 9E 05 0C
10930: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01
10940: 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00
10950: 11 00 13 3C 3C 0E 00 1B 4A 0D 00 23 57 12 00 2B
10960: 69 1E 00 33 87 2E 00 3B B5 54 00 42 FF 00 00 3C
10970: FF 00 00 3C FF 00 00 3C FF 00 00 00 00 00 00 00
10980: 12 00 14 3C 3F 28 00 1E 6F 28 00 28 9F 24 00 2D
10990: B7 24 00 32 CF 24 00 37 E7 24 00 3C FF 00 00 3C
109A0: FF 00 00 3C FF 00 00 3C FF 00 00 00 00 00 00 00
109B0: 18 00 14 3C 3F 28 00 1E 6F 28 00 28 9F 24 00 2D
109C0: B7 24 00 32 CF 24 00 37 E7 24 00 3C FF 00 00 3C
109D0: FF 00 00 3C FF 00 00 3C FF 00 00 00 00 00 00 00
109E0: 00 00 00 00 00 00 02 32 00 31 00 31 00 31 00 31
109F0: 00 00 00 31 00 32 00 38 00 00 00 00 00 00 00 00
```
The value at 10947 looks suspiciously like the flag you mean to disable.
I haven't been able for the life of me to find out just *how* HWiNFO64 manages to access this dump:

I tried accessing RAM using the Super I/O index/data interface on ports 0x4E/0x4F, LDN=0x0f (SMFI) - needed to be mapped to an IO port space first as it was uninitialized, but I was only able to access the firmware flash for the "IT87952" (which interestingly reads "IT873x" in the signature string location):

[image]

Tried various of the higher order address bits in IO+2/IO+3 but besides the firmware flash there was only FF's...

I then proceeded to try accessing via the power management interface (LDN=0x11 (PMC1)) which is basically a Host-EC bus bridge. Supposedly the firmware has an interface to read and write memory and registers via a the legacy EC ACPI interface on ports 62h (PMDIR/PMDOR) and 66h (PMCMDR/PMSTR). The low-level interface responded as expected with the status flags while programming the bus read requests but the firmware ultimately didn't react. I got the information about the "high-level" firmware interface from the `it_ecfw.c` source file of an unpublished branch of the flashrom project (coreboot sub-project): https://review.coreboot.org/c/flashrom/+/55715/30/ite_ecfw.c#199

Just wanted to drop these bits of info here; maybe I'm just missing a tiny piece of information and someone else has a clue. The ultimate goal would be:
 - to get the EC to enable its Host bus mapping so the EC RAM can be MMIO'd
 - or find a way to read/write EC RAM without MMIO
 - or find out how HWiNFO64 manages to get access to the data in question.

[78 more replies]

PR #1020 (2023-03-21): Add support for Gigabyte's custom fan controller
* This is a custom firmware loaded into ITE ECs that adds automatic control of fans, and possibly LEDs and other functions as well.
* It has only been seen so far in the 2nd EC of some MBs, and requires the ITE chip to support reading its "scratch ram".
* The controller fan control module is disabled when attempting to change the fan speed for the known affected fans.
* Also clean up the discovery and name of some of the ITE chips based on known info.
* Only AMD MBs are supported with this commit, further work required for Intel boards.
* Fixes  [... cut]