LibreHardwareMonitor/LibreHardwareMonitor #252: ITE IT8665E Reads voltages wrongly

DjordjeMandic, 2020-05-21:
Compared to OpenHardwareMonitor there is big difference in VBat voltage readout and there is also few voltages missing.
[image]
Other voltages seem fine, little difference. Changing resistor divider ratio to 10 10 from 0 1 make readings a little bit okay but still not the same. 

Question: Are you doing development for this fork independently of ohm or youre following its progress and importing the changes from official ohm?

Phyxion, 2020-06-02:
Can you check the latest build again?

Phyxion, 2020-06-03:
@DjordjeMandic Please give this a try:
[Release.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/4725908/Release.zip)
Write these files over the latest build on main page.

DjordjeMandic, 2020-06-04:
[image]
Yeah its better now but i still dont get it why is LHM measuring little bit less compared to OHM, look at maximum values. VBat is still off

closed 2020-06-05 by Phyxion

Phyxion, 2020-06-05:
That's because OHM has a different voltage gain for it, [IT8665E ](https://github.com/a1wong/it87/blob/master/it87.c#L718)uses 0.0109 gain and not 0.011.

commit fb58f13c (2020-06-05): Change ITE configurations

Fixes #252.