LibreHardwareMonitor/LibreHardwareMonitor #257: Gets stuck on start

Maxus1998, 2020-05-22:
Every time I start Libre Hardware Monitor and click on the try icon to get the window it seems like it is stuck.
I can't scroll or click anything, not event "File", "View" etc.
I also can't click the right click options in the tray icon ("Hide/Show" or "Exit").

Only way to close it is via killing the process in task manager.

When I call the API in python with " wmi.WMI(namespace="root\LibreHardwareMonitor")" I get some sensors, even tough most of them appear only when it is running for more than a minute.

Are there any logs to see what the tool is currently doing / if it is stuck on initializing some part of my hardware?

DjordjeMandic, 2020-05-23:
Im having similar problem, but in my instance when i open it its just frozen for some time then numbers start updating really fast then its all just back to normal. Have you tried waiting, maybe it responds? I usually wait 10-15 seconds.

Maxus1998, 2020-05-23:
I just let it running for a while and it seems like after some minutes it starts working like expected

DjordjeMandic, 2020-05-23:
@Maxus1998 yeah same problem

SkyEmie, 2020-08-05:
Same problem here too

closed 2025-12-30 by PhyxionNL