LibreHardwareMonitor/LibreHardwareMonitor #265: IT8XXX missing controls

Rem0o, 2020-06-01:
https://github.com/openhardwaremonitor/openhardwaremonitor/commit/7bf4f655d2c3ad9b660b6703c632b1d5694778ca
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/1104b661304354c868f4f97554e854058841c094

I'm comparing both these commits and it seems like some parts were lost in translation. More specifically, "the switch (chip)  case Chip.IT8665E:" to create the Control array is different. Well in fact, the whole switch statement seems to have a few differences. Tested the 5 lenght Control array and it works for the IT8665E.

TheTrueBumba, 2020-06-01:
Since you made a pull request, take a look at my post. LHW doesnt detect IT8655E and I had to change the vendor ID, but I dont know the impact to others users.
Link to my issue request. If I am correct, you will need to make a new pull request for IT8655E:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/246

Phyxion, 2020-06-01:
I've combined some checks as there were a lot of duplicate calls in original commit. Do you have any specific lines that are missing @Rem0o? I can take a look later.

Rem0o, 2020-06-01:
@Phyxion  line 211 of https://github.com/openhardwaremonitor/openhardwaremonitor/commit/7bf4f655d2c3ad9b660b6703c632b1d5694778ca

Or overall, this commit added 2 new control register in the FAN_PWM_CTRL_REG array for specific chips. These need to be accounted for in all the different methods that itterate with an index over the controls.

TheTrueBumba, 2020-06-02:
Dear friends, take a look in the line below. The line *breaks* support for IT8655E.

if (!valid || vendorId != ITE_VENDOR_ID)
return;

The two constants doesnt match and always return and break the sensor check for ASUS PRIME B450M-GAMING/BR

I had to change my ITE_VENDOR_ID to 0x7F

Results after the change:

[image]

closed 2020-06-02 by Phyxion

oxirt, 2020-11-08:
Please note that the fan readings are also incorrect. See ticket 277.

commit 5976a5a7 (2020-06-02): Update IT87XX.cs

Fixes #265
Fixes #246