LibreHardwareMonitor/LibreHardwareMonitor #266: Exception when Laptop goes to Standby

Bluejanis, 2020-06-05:
I see this Exception, when I come back to my Laptop after a while:

> System.InvalidOperationException: Die Sammlung wurde geändert. Der Enumerationsvorgang kann möglicherweise nicht ausgeführt werden.
>    bei System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
>    bei System.Collections.Generic.List`1.Enumerator.MoveNextRare()
>    bei LibreHardwareMonitor.Hardware.Computer.Traverse(IVisitor visitor)
>    bei LibreHardwareMonitor.UI.MainForm.Timer_Tick(Object sender, EventArgs e)
>    bei System.Windows.Forms.Timer.OnTick(EventArgs e)
>    bei System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
>    bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

It is reproducable on my machine. So basically when I leave the laptop alone for a bit and some energy saving gets enabled.

I had the graph (plot), the main window and 2 tray temperature icons active.

I'm sorry that Microsoft shows the Exception in German language. It sounds like a collection got changed while iterating over it.

tacal666, 2020-06-07:
same issue after sleep or hibernate

-------------------------------------------------------------------------------------------------------
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.InvalidOperationException: Collection was modified; enumeration operation may not execute.
   at System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   at System.Collections.Generic.List`1.Enumerator.MoveNextRare()
   at LibreHardwareMonitor.Hardware.Computer.Traverse(IVisitor visitor)
   at LibreHardwareMonitor.UI.MainForm.Timer_Tick(Object sender, EventArgs e)
   at System.Windows.Forms.Timer.OnTick(EventArgs e)
   at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
-------------------------------------------------------------------------------------------------------

skobkin, 2020-06-09:
Same for me on ASUS Zephyrug G15.

```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.InvalidOperationException: Collection was modified; enumeration operation may not execute.
   at System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   at System.Collections.Generic.List`1.Enumerator.MoveNextRare()
   at LibreHardwareMonitor.Hardware.Computer.Traverse(IVisitor visitor)
   at LibreHardwareMonitor.UI.MainForm.Timer_Tick(Object sender, EventArgs e)
   at System.Windows.Forms.Timer.OnTick(EventArgs e)
   at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4180.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
LibreHardwareMonitor
    Assembly Version: 0.8.1.0
    Win32 Version: 0.8.1.0
    CodeBase: file:///C:/ProgramData/chocolatey/lib/libre-hardware-monitor/tools/LibreHardwareMonitor.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Aga.Controls
    Assembly Version: 1.7.0.0
    Win32 Version: 1.7.0.0
    CodeBase: file:///C:/ProgramData/chocolatey/lib/libre-hardware-monitor/tools/Aga.Controls.DLL
----------------------------------------
LibreHardwareMonitorLib
    Assembly Version: 0.8.1.0
    Win32 Version: 0.8.1.0
    CodeBase: file:///C:/ProgramData/chocolatey/lib/libre-hardware-monitor/tools/LibreHardwareMonitorLib.DLL
----------------------------------------
Microsoft.Win32.TaskScheduler
    Assembly Version: 2.8.15.0
    Win32 Version: 2.8.15.0
    CodeBase: file:///C:/ProgramData/chocolatey/lib/libre-hardware-monitor/tools/Microsoft.Win32.TaskScheduler.DLL
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4180.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
[... 27 lines cut ...]
System.Threading
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Threading/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Threading.dll
----------------------------------------
System.Collections
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Collections/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Collections.dll
----------------------------------------
System.Configuration.Install
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration.Install/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.Install.dll
----------------------------------------
uad5edsr
    Assembly Version: 0.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
HidSharp
    Assembly Version: 2.1.0.0
    Win32 Version: 2.1.0.0
    CodeBase: file:///C:/ProgramData/chocolatey/lib/libre-hardware-monitor/tools/HidSharp.DLL
```

Eno77, 2020-06-12:
Same here!

Additional information:
In the Sensor Treeview (and in the gadget) the LAN related nodes are gone after this error.
So probably has to do something with the network device list.

LiloBzH, 2020-06-14:
Same here https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/272

tlusk, 2020-06-20:
This is most likely caused by 738752b8999913fcd02956124d50eb410db4178d

@Phyxion @moellerm

giddyhup, 2020-07-10:
I have the same error but actually the applications seems to continue working normally. LAN metrics are being displayed.

SkyEmie, 2020-07-16:
Same error here :/

tlusk, 2020-07-16:
FYI this is the most recent build that doesn't have this problem: https://ci.appveyor.com/project/LibreHardwareMonitor/librehardwaremonitor/builds/33096165/artifacts

Eno77, 2020-07-16:
unfortunately the linked version is not the most recent build. It goes to 1.0.194 - most recent (with bug) is 1.0.204

https://ci.appveyor.com/project/LibreHardwareMonitor/librehardwaremonitor/history

tlusk, 2020-07-16:
Correct, every version after 1.0.194 contains this bug, so if you want a workaround to avoid it for the time being you can use the older 1.0.194 version.

Phyxion, 2020-07-17:
This is indeed caused by the power change, but it could've happen in some other cases as well. I've committed a fix for it.

Eno77, 2020-07-21:
Thanks Phyxion.
Unfortunately I must report it is not fixed completely yet.

The Error does still happen and an according Error/Crash Window opens after wakeup.
But the Network nodes are not gone after this and everything seems to work normal.
So it seems everything is handled correctly and left is just to prevent the crash window from opening.

Phyxion, 2020-07-21:
Is the error still the same as above? I don't receive it anymore on my machine.

Eno77, 2020-07-21:
Seems so:

`************** Ausnahmetext **************
System.InvalidOperationException: Die Sammlung wurde geändert. Der Enumerationsvorgang kann möglicherweise nicht ausgeführt werden.
   bei System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   bei System.Collections.Generic.List`1.Enumerator.MoveNextRare()
   bei LibreHardwareMonitor.Hardware.Computer.Traverse(IVisitor visitor)
   bei LibreHardwareMonitor.UI.MainForm.Timer_Tick(Object sender, EventArgs e)
   bei System.Windows.Forms.Timer.OnTick(EventArgs e)
   bei System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)`

giddyhup, 2020-07-21:
I still get the error (version 0.8.1.0).

Phyxion, 2020-07-21:
Please try the latest build, I managed to get the exception now also on my machine, it didn't always occur.

Eno77, 2020-07-21:
thanks for the quick fix, I will test tomorrow.

Meanwhile I commented on line 374 - there is a "remove" outside a lock, which seems suspivious to me and may be the real root cause?

Phyxion, 2020-07-21:
Remove itself locks.

LiloBzH, 2020-07-21:
it seem to be ok after the last patch. No crash after sleep mode for me :-) Testing in progress

Eno77, 2020-07-24:
Yes, after these few days I did not have any issue anymore. Thanks for the fix again!

Still weird, that you cant use the foreach there - would be a nice thing to investigate, maybe I pull the projekt again some day and have a look wehre the change happens / may happen.
But so far everything looks OK now.

Brittank88, 2022-04-07:
Forgive me for reviving an old issue but I am still experiencing this problem (LHM v0.8.9 on Win10 Pro 21H1). Should I be creating a new issue, or is it appropriate to continue here?

riQQ, 2022-04-07:
See #385

commit 9454fab5 (2020-07-21): Fixes #266